arduino/ArduinoCore-mbed #889: missing null pointer checks

ddmesh, 2024-05-27:
https://github.com/arduino/ArduinoCore-mbed/blob/dd5abb6171063b2777620b9339510c63b915ce72/libraries/Ethernet/src/Ethernet.cpp#L6

Per default an object is created via `arduino::EthernetClass Ethernet;`. In this case the `eth_if` member got initialized (which is a pointer).

But when user creates his own object and passes a null pointer for EthernetInterface()
`arduino::EthernetClass Ethernet(nullptr);`
Several calls to other public functions, **will directly call/access the null pointer**.

https://github.com/arduino/ArduinoCore-mbed/blob/dd5abb6171063b2777620b9339510c63b915ce72/libraries/Ethernet/src/Ethernet.cpp#L17
https://github.com/arduino/ArduinoCore-mbed/blob/dd5abb6171063b2777620b9339510c63b915ce72/libraries/Ethernet/src/Ethernet.cpp#L56
https://github.com/arduino/ArduinoCore-mbed/blob/dd5abb6171063b2777620b9339510c63b915ce72/libraries/Ethernet/src/Ethernet.cpp#L71
https://github.com/arduino/ArduinoCore-mbed/blob/dd5abb6171063b2777620b9339510c63b915ce72/libraries/Ethernet/src/Ethernet.cpp#L80

manchoz, 2024-06-11:
Hey, @ddmesh, would you mind adding the null-pointer checks where needed and submitting a PR?

Thanks!

maidnl, 2024-06-17:
@ddmesh thank you for your interest in reporting potential problems, however the scenario you described is impossible to achieve. If you try to compile something with `arduino::EthernetClass Ethernet(nullptr);` you'll get a compiler error, this means that eth_if is always initialized (`EthernetInterface *eth_if = &net;`). However I agree that is a bit confusing so I made a PR to properly check if eth_if is nullptr, just for clearness.

PR #901 (2024-06-20): Ethernet, initCallback, ctor and eth_if
Possible alternative to #897 #898 #900 it adds function pointers cleanup and uses the same ctor logic of WiFi.

Opened as a draft: it should build but is untested

Fixes #890 
Fixes #889
closed 2024-06-20 by pennam