arduino/ArduinoCore-mbed #916: `Ethernet.dnsServerIP()` does not return the IP address I set with `Ethernet.begin()`

haveyouseenf, 2024-07-03:
While working with the Ethernet library on my Opta I stumbled upon this weird behavior. You can easily reproduce the issue with the following sketch:

```cpp
#include <Ethernet.h>

IPAddress ip(192, 168, 10, 15);
IPAddress subnet(255, 255, 255, 0);
IPAddress dns(1, 1, 1, 1);
IPAddress gateway(192, 168, 10, 1);

void setup() {
  Serial.begin(9600);

  Ethernet.begin(ip, dns, gateway, subnet);

  Serial.print("IP = ");
  Serial.print(Ethernet.localIP());
  Serial.print(", DNS server = ");
  Serial.print(Ethernet.dnsServerIP());
  Serial.print(", Default gateway = ");
  Serial.println(Ethernet.gatewayIP());
}

void loop() {
}
```

On the serial monitor I get the following output:

```text
IP = 192.168.10.15, DNS server = 8.8.8.8, Default gateway = 192.168.10.1
```

For some reason the DNS server always returns `8.8.8.8`. If I can provide further details let me know!

JAndrassy, 2024-07-03:
what version of the Arduino Mbed platform do you have installed?

haveyouseenf, 2024-07-03:
I'm using `mbed_opta` version `4.1.3`.

closed 2024-09-09 by facchinm