arduino/ArduinoCore-mbed #919: Can't have input A0 and A3 configured as interrupt on Opta

bcdrme, 2024-07-05:
Configuring both A0 and A3 with `digitalPinToInterrupt` will crash on Arduino Opta (tested on 3 devices so far, it crashes 100% of the time).

Commenting out either A0 or A3 in the example code below and it will run fine.

Example code :
```C++
#include <Arduino.h>

void ISR() {}

void setup() {
  Serial.begin(9600);
  while (!Serial)
    ;

  delay(2000);
  Serial.println("Can't have A0 and A3 as interrupts.");
  delay(2000);

  attachInterrupt(digitalPinToInterrupt(A0), ISR, CHANGE);
  attachInterrupt(digitalPinToInterrupt(A1), ISR, CHANGE);
  attachInterrupt(digitalPinToInterrupt(A2), ISR, CHANGE);
  Serial.println("digitalPinToInterrupt(A3) will fail if A0 is not commented out.");
  attachInterrupt(digitalPinToInterrupt(A3), ISR, CHANGE);
  Serial.println("A3 OK.");
  attachInterrupt(digitalPinToInterrupt(A4), ISR, CHANGE);
  attachInterrupt(digitalPinToInterrupt(A5), ISR, CHANGE);
  attachInterrupt(digitalPinToInterrupt(A6), ISR, CHANGE);
  attachInterrupt(digitalPinToInterrupt(A7), ISR, CHANGE);
}

void loop() {
  Serial.println("Looping...");
  delay(2000);
}
```

manchoz, 2024-07-09:
@bcdrme, this behavior is documented in the user manual at https://docs.arduino.cc/tutorials/opta/user-manual/#interrupts.

closed 2024-07-09 by facchinm

bcdrme, 2024-07-09:
@manchoz @facchinm 

Thank you, there is a mistake though : 

> Due to Opta's microcontroller interrupt structure, interrupts in terminals I1 (A0) and I4 (A4) cannot be used simultaneously to avoid operational issues. It is important to note that, despite this limitation, any other combination of inputs can be used for interrupt detection. However, this means that, at most, seven of the eight available inputs can be used simultaneously for interrupts, as combinations containing both I1 and I4 are excluded from viable configurations.

> I1 (A0) and I4 (A4)
A3 not A4. 🙏

jcarolinares, 2024-07-15:
Hello @bcdrme thank you so much for the issue, regarding documentation the team is working on it and will be fixed soon, we will ping you when it is done!

jcarolinares, 2024-07-17:
Hello @bcdrme again for your feedback pointing out this error. The related documentation has been double checked and updated and it will be publicly update pretty soon.