arduino/ArduinoCore-mbed #920: Giga R1 Wifi - Wifi-Access-Point only Possible with Secrets of existing WiFi-Network

Sternie73, 2024-07-09:
Hello,

at first, sorry for my bad english, I'm from Germany.

I try to open an Access-Point with the example "Web Server AP Mode" form the [Arduino-Tutorials](https://docs.arduino.cc/tutorials/giga-r1-wifi/giga-wifi/#web-server-ap-mode) for the Giga R1 Wifi.

With the SSID and Password of my existing Network there will be created an Access-Point.
When I change the SSID or Password, the creating failed.

What can I do?

pennam, 2024-07-17:
Hello @Sternie73,
this is very strange, i've tested your same example with my Giga R1 WiFi and i'm able to create and AP with the following credentials

SSID:`test`
PWD:`12345678`

Is the antenna mounted on your board?

giorgiogallina, 2024-09-06:
I had issues as well opening access point.
It resulted that the password has to be at least 8-characters long (so, for instance, neither "qwerty" nor "qwerty1" are accepted).

By the way, is there any documentation where we can find information on how to best use WiFi library on GIGA R1? That would have saved me one hour of debuging trying to guess what was wrong with my attempt to create an Access Point (and many many more hours that I am still "wasting")

Thank you

megacct, 2024-09-06:
This PR #826 made beginAP return an error rather than crash the OS (took me much longer than an 1hr to find)