arduino/ArduinoCore-mbed #922: mbed EoL

sstaub, 2024-07-11:
When will Arduino use Zephyr instead of mbed? End of Life is announced for mbed platform.

savejeff, 2024-07-12:
I recommend switching to the [earlephilhower Core](https://github.com/earlephilhower)
it is already more feature-rich and supports/implements a full rtos as far as I know.

sstaub, 2024-07-12:
For Raspberry Pi Pico I use that core but for all the STM32 stuff?

savejeff, 2024-07-12:
Ah right, the mbed core is now also used for the STMs. 
just out of curiosity: what about the stm32duino Core? it doesn't support the Arduino STM boards like the Mega right?

JAndrassy, 2024-07-12:
Arduino Mbed Core uses mbed-os version from 2019. how can EOL of Mbed development have any influence on the Arduino Mbed Core?