arduino/ArduinoCore-mbed #924: CAN.write() return-code does not comply to API spec

samuelsadok, 2024-07-19:
`CAN.write()` in this core returns 0 for errors but should return a negative error code.

### Details

The docstring for the Arduino CAN API [states](https://github.com/arduino/ArduinoCore-API/blob/master/api/HardwareCAN.h#L90) that `HardwareCAN::write()` returns:
> 1 if the message was enqueued, an _implementation defined_ error code < 0 if there was an error

This core implements `HardwareCAN::write()` by calling `mbed::CAN::write()` and directly returning the result ([code](https://github.com/arduino/ArduinoCore-mbed/blob/1399d6477548f3f5585155652511d68d3ef85951/libraries/Arduino_CAN/src/Arduino_CAN.cpp#L60), [header](https://github.com/arduino/ArduinoCore-mbed/blob/1399d6477548f3f5585155652511d68d3ef85951/libraries/Arduino_CAN/src/Arduino_CAN.h#L64)).

However `mbed::CAN::write()` has a [different spec](https://os.mbed.com/docs/mbed-os/v6.16/mbed-os-api-doxy/classmbed_1_1_c_a_n.html#aabf4ced53511675be2b61c959b681519) for returning errors:
> 0 if write failed, 1 if write was successful

### Discussion

This was brought up by @bthacher-crabi in https://github.com/odriverobotics/ODriveArduino/issues/4 and I'd be interested what the maintainers of this repo think. Technically the meaning of `return 0` is unspecified in the API, so it's unclear what client code should do with it. It's conceivable that other cores use 0 to mean "success" (which is a fairly common convention). Should we add a preprocessor special case for the mbed core specifically?

Separately, going forward it would be cool if either the code or the API specs are changed so that they are consistent.

samuelsadok, 2024-09-16:
Any comment on this? (perhaps @facchinm?)

aentinger, 2024-09-16:
Hi @samuelsadok :coffee: :wave: Thank you for reporting this, you are indeed correct. I'll prepare a PR with a fix.

PR #956 (2024-09-20): Fix: Arduino_CAN::write return value incompatible with interface specification at ArduinoCore-API/HardwareCAN.
This fixes #924.
commit 931f0bbe (2024-09-20): Fix: Arduino_CAN::write return value incompatible with interface specification at ArduinoCore-API/HardwareCAN. (#956)

This fixes #924.
closed 2024-09-20 by aentinger