arduino/ArduinoCore-mbed #931: A9 analogRead(PureAnalogPin) issue with M4 core

white-ar, 2024-07-26:
analogRead(A9) generates an undefined reference to `analogRead(PureAnalogPin)' error when compiled for the M4 core.
Setup: Giga R1 board library: 4.1.5

Sketch:

```
#include "Arduino.h"
#include "RPC.h"

#define SERVO_POT_1     A9

void setup() {
  RPC.begin();
  analogReadResolution(16); //GIGA has 16 bit ADC  (0-65535)  
}

void loop() {
int temp = analogRead(SERVO_POT_1);
}

```

Error:

Error message:
C:...\AppData\Local\Temp\arduino\sketches\D2129B132D44D2F5F57EC30CA80C2E2B\sketch\sketch_jul25a.ino.cpp.o: In function loop': C:\...\AppData\Local\Temp\.arduinoIDE-unsaved2024625-43476-ja9cee.pq8d\sketch_jul25a/sketch_jul25a.ino:20: undefined reference to A9'
C:...\AppData\Local\Temp.arduinoIDE-unsaved2024625-43476-ja9cee.pq8d\sketch_jul25a/sketch_jul25a.ino:20: undefined reference to `analogRead(PureAnalogPin)'
collect2.exe: error: ld returned 1 exit status

Using library RPC at version 1.0 in folder: C:...\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.1.5\libraries\RPC
Using library rpclib at version 1.0.0 in folder: C:...\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.1.5\libraries\rpclib
Using library openamp at version 1.0 in folder: C:...\AppData\Local\Arduino15\packages\arduino\hardware\mbed_giga\4.1.5\libraries\openamp_arduino
exit status 1

Compilation error: exit status 1

Any ideas?

njadNissi, 2024-10-16:
All the analog pins from A8-A11 are called PureAnalogPin, and they have undefined reference pin numbers under "pins_arduino.h" file. I am encountering the same problem too. check my issue #972.

njadNissi, 2024-10-16:
so you know the problem is valid for both cores and any of the 4 pins from A8-A11