arduino/ArduinoCore-mbed #937: Every 5th TCP connection fails on Arduino Giga R1.

schnoberts1, 2024-08-15:
Platform: Giga R1
Arduino Core Mbed 4.1.5
HttpClient: 0.6.1

Here's a sequence of events with a crudely instrumented MbedClient. We connect using ArduinoHttpClient every 1s. Every 5th connection fails with error -3005 from `static_cast<TCPSocket *>(sock)->open(getNetwork());` in the `MbedClient::connect() call`. When it fails it takes 30s to return. This is in version 4.1.5. In version 4.1.1 it would never recover and continue to fail until device reboot. There would be no 30s delay. This issue occurs on any http server I've tried. I can continue to connect to the server in the code with python on my Mac with no problems. If I extend the delay so I connect to the server once every 5s the problem goes away. At 4s it reappears. 

```
22:10:12.229 -> Fetching content length from httpforever.com:80
22:10:12.393 -> Return code from connect0
22:10:13.213 -> took=993ms length: 5124
22:10:15.229 -> Fetching content length from httpforever.com:80
22:10:15.394 -> Return code from connect0
22:10:16.216 -> took=979ms length: 5124
22:10:18.198 -> Fetching content length from httpforever.com:80
22:10:18.365 -> Return code from connect0
22:10:19.225 -> took=1008ms length: 5124
22:10:21.203 -> Fetching content length from httpforever.com:80
22:10:21.366 -> Return code from connect0
22:10:22.222 -> took=1004ms length: 5124
22:10:24.202 -> Fetching content length from httpforever.com:80
22:10:24.202 -> Return code from TCPSocket::open-3005
22:10:54.295 -> took=30099ms length: -1
```

code

```
#include <Arduino.h>
#include <WiFi.h>
#include <HttpClient.h>

void setup() {
  Serial.begin(115200);
  delay(1000);
  Serial.println("HI");
  // put your setup code here, to run once:
  auto wifi = WiFiInterface::get_default_instance();
  wifi->set_dhcp(true);
  wifi->set_credentials("ZZZZ", "ZZZZ", nsapi_security_t::NSAPI_SECURITY_WPA2);
  Serial.println(wifi->connect());
}

void loop() {
  // put your main code here, to run repeatedly:
  WiFiClient client;
 
   HttpClient httpClient(client, "httpforever.com", 80);
   Serial.println("Fetching content length from httpforever.com:80");
   auto then = millis();
   httpClient.beginRequest();
   httpClient.get("/");
   httpClient.endRequest();
   auto len = httpClient.contentLength();
   Serial.print("took="); Serial.print((millis() - then)); Serial.print("ms "); Serial.print("length: "); Serial.println(len);
   delay(5000);
}
```

schnoberts1, 2024-08-15:
... also works fine at 4.5s delay

schnoberts1, 2024-08-15:
... I am also guessing this 30s timeout is in HttpClient since there's no delay on the return from the failed connect.

schnoberts1, 2024-08-16:
Note: calling client.stop() does not resolve the issue

stopandchatchfire, 2024-09-05:
The maximum number of TCP connections is 4.

As defined in the file: "mbed_config.h"

`#define MBED_CONF_LWIP_TCP_SOCKET_MAX    4`

schnoberts1, 2024-09-05:
I never have 4 tcp sockets open but I am confident this limit is at the core of the issue. Some clean up isn’t being executed.

JAndrassy, 2024-09-09:
@schnoberts1 can you try these changes? https://github.com/arduino/ArduinoCore-mbed/pull/912/files

pennam, 2024-10-01:
@schnoberts1 the 30s timeout comes from the Http library, you can get rid of it adding `httpClient.setHttpResponseTimeout(1000);` to your sketch.

I'm still investigating about the `NSAPI_ERROR_NO_SOCKET` return value but i agree it is related to the `MBED_CONF_LWIP_TCP_SOCKET_MAX` define and in the way lwip TCP connections are deleted https://github.com/arduino/mbed-os/blob/e0cad5c4277b3c5ee9cb01a9529df9333612bbeb/connectivity/lwipstack/lwip/src/api/lwip_api_lib.c#L233

It looks that before complete deletion they can be put in a waitstate and closed afterwards

https://github.com/arduino/mbed-os/blob/e0cad5c4277b3c5ee9cb01a9529df9333612bbeb/connectivity/lwipstack/lwip/src/api/lwip_api_msg.c#L913

JAndrassy, 2024-10-01:
does the HttpClient send the "Connection: close" header?

pennam, 2024-10-01:
@JAndrassy  yes HttpClient sends the "Connection: close" header. 

I think I've found the rootcause, and it is server side, in fact `httpforever.com` keeps sending  `TCP Keep-Alive` messages, and thus connections are not closed. Using a different server like `example.org` connections are closed correctly and the `NSAPI_ERROR_NO_SOCKET` error is not generated..

schnoberts1, 2024-10-01:
Good find! We close the socket though so I would have thought LWIP would have sent FIN and either got a FIN ACK and closed the underlying socket or timed out waiting for one. I wonder, is LWIP not sending FIN?

pennam, 2024-10-01:
Here is what i get with wireshark, I'm now thinking if the TCP Window Full messages are somehow involved
[wireshark.zip](https://github.com/user-attachments/files/17203040/wireshark.zip)

pennam, 2024-10-01:
I get the same `NSAPI_ERROR_NO_SOCKET` using anothe http site `sneaindia.com` `TCP Window Full`
 messages are also present.
[snea.pcapng.zip](https://github.com/user-attachments/files/17203234/snea.pcapng.zip)

schnoberts1, 2024-10-01:
Interesting. I can see on the exampleorg PCAP that the Giga sends a GET and then it gets the data and an HTTP 200 response followed immediately by a FIN. The Giga then sends an ACK for data up to sequential 537 and then an ack for data up to 1619 (which includes the FIN that got sent to the Giga). The Giga never sends a FIN in that packet capture. It should do that. I think the server is then in FIN_WAIT2 but since the HTTP server has closed the socket it’ll time it out. 

In the httpforever it gets interesting. 

The Giga sends the GET (split across two packets weirdly). Packet 24 has the GET and packet 25 has / HTTP/1.1 Host: httpforever.com
The same happens again, the Giga sends User-Agent in one packet then : Arduino/2.2.0 \nConnection: close in the other. 

At this point it’s the same as the exampleorg capture really. A key difference is that the exampleorg payload is less than the advertised window (2144) so it never needs to write a packet with WindowFull set.

The server sends the data with the Connection: close also in the header. Not all the data is in the packet so the next packet is more of the data but the TCP Window Full is set.
This is because the advertised window from the Giga is 2144 and it’s sent exactly 2144 bytes across those two packets. It needs an ACK now.
The Giga ACKs the two packets.
The server sends a 536 byte packet so it can’t be finished, the content length was around 5k. The Giga acks it but now its window is down to 536 bytes. 
The server fills the window and tags the packet accordingly.
The Giga announces a zero window (I don’t know why, it’s ACKed everything - but then I am not a TCP guru).
The giga then RST’s the connection. At this point it should be closing the underlying socket since it’s an abort. It does not need to wait for anything from the server. However, it doesn’t abort. The server sends a TCP keep alive after the RST and the Giga responds to it! I don’t think that is right. 

You are right, it could be the WindowFull but the response from the Giga seems odd at this stage. I do not understand why it continues to process responses after sending a RST.

pennam, 2024-10-01:
Increasing TCP MSS to 1460 seems to fix this issue using `httpforever.com`

schnoberts1, 2024-10-01:
Does it result in a larger advertised window? Do the WIndowFull flags go away?

pennam, 2024-10-01:
Here it is the capture with increased MSS, no `TCP Window Full` occurs
[mss-increased.pcapng.zip](https://github.com/user-attachments/files/17209636/mss-increased.pcapng.zip)

schnoberts1, 2024-10-01:
Doesn’t this mean the increase to MSS is simply hiding an underlying problem?

schnoberts1, 2024-10-01:
I will take a look at it tomorrow as well

megacct, 2024-10-01:
FYI, In addition to the mbed conf settings mentioned here https://github.com/arduino/ArduinoCore-mbed/pull/966#issuecomment-2385659878, I've also set
`compiler.c.extra_flags=-DSOCKET_BUFFER_SIZE=2048`
`compiler.cpp.extra_flags=-DSOCKET_BUFFER_SIZE=2048`
in my platform local

megacct, 2024-10-01:
Hi @schnoberts1, I ran your sketch on my setup for 5mins with 5 sec delay, 5mins with 1 sec delay and for 2mins with 100ms delay - output attached.

schnoberts1, 2024-10-01:
This makes sense as I think the increased MSS results in the issue being masked.

megacct, 2024-10-01:
Just re-ran your sketch (even thoguh after bedtime) with stock mbedlib and just the SOCKET_BUFFER_SIZE override. It works. Please try

schnoberts1, 2024-10-01:
That’s a good find. I will try tomorrow. I bet this is because with the extra socket buffer wise it’s not advertising a smaller TCP window because it’s for room for the payload . Thank you very much. I think I will spin up a local web server and test it against payload size.

pennam, 2024-10-02:
just increasing `SOCKET_BUFFER_SIZE` seems to not fix the issue in my case

megacct, 2024-10-02:
My apologies, you are correct. It must not have used the stock lib (it was late)

schnoberts1, 2024-10-02:
I looked at the packet capture (increased MSS). The increased MSS size results in a larger window being advertised by the Giga. It advertises a 5840 byte window which is larger than the payload that's being sent to it. We get no ZeroWindow. It also means that the http server can send the last bit of data with the FIN state set so when the RST/ACK is sent the socket is already presumably half closed as the Giga got a FIN.

Without this increase the server can't send all the data, the FIN state is not sent, the RST/ACK is sent on an open connection. The Server doesn't abort its side of the connection and sends Keep Alives (which I don't think is right, but the TCP stack is supposed to behave well when things aren't quire right). The Giga's stack is processing these keep alives despite the fact it's aborted the connection. I don't think this is correct. The result is the socket being held open incorrectly. I propose this is the issue and what we see with increased MSS (etc) is a mask. I can't build a custom mbed build at the moment, but I imagine testing this hypothesis would involve an http server with keep-alive that responds like the one I chose, but sends a much larger http body content, triggering the zero window, no FIN etc. 

To sort of demonstrate this, if I don't just call contentLength() but instead drain the connection by calling responseBody() then the problem goes away, because the FIN can be sent, RST is on a half closed socket and the Keep Alives aren't set.

pennam, 2024-10-02:
> Does that make sense?
it makes perfectly sense, I've done other tests with the increased MSS and trying to download bigger files... know what? the issue comes back again, so i agree with you increasing MSS is masking the issue and is not the fix.

schnoberts1, 2024-10-02:
I think at this stage it might be worth confirming with the LWIP crowd what they expect to happen in the situation given the PCAP showing the failure mode. Maybe I am wrong and this is the expected behaviour and it's fine because there's a timeout that handles it eventually.

To some extent I think you could triage this issue away if you were inclined to on the basis of:

1. The connection is torn down eventually.
2. Always read the entire stream from the server and if you don't then don't expect the stack to free resources in a timely fashion. 
3. It is a somewhat contrived example. 

... but it'd be nice to get a response from the LWIP community (I assume this is where the issue might lie, or perhaps not.... skimming the LWIP code it does look like resource freeing happens on a RST and also shortly after in its background thread that cleans up dead connections - that's an every 0.5s/0.25s interval thread though, not a 5s interval).

I should probably change the title of this ticket to be clearer about the situation.

BTW: Obv when I fully drain the connection by calling responseBody() on HttpClient the problem doesn't occur.

schnoberts1, 2024-10-02:
... I guess the counter argument is that there are plenty of situations where you might want to abandon your connection while the server is still writing data.

schnoberts1, 2024-10-02:
Also, I do appreciate the effort put into looking at this. It's helped a lot.