arduino/ArduinoCore-mbed #941: Feature request - Implement Serial.printf()

rei-vilo, 2024-08-17:
Some Arduino Nano boards implements `Serial.printf()` but not all.

+ The Nano RP2040 Mbed-OS and Nano BLE Mbed-OS do not support `Serial.printf()`.
+ The Nano ESP32 and Nano Matter support `Serial.printf()`.

Please implement `Serial.printf()` on Arduino Core-Mbed to ensure compatibility across all the Arduino Nano family.

Thank you!

per1234, 2024-08-17:
Hi @rei-vilo. This would need to be implemented in the arduino/ArduinoCore-api codebase (which is a dependency of this project), not in this repository's codebase.

The request is already tracked in the appropriate issue tracker: https://github.com/arduino/ArduinoCore-API/pull/28

So you can go ahead and close this as duplicate.

rei-vilo, 2024-08-19:
Thnak you! I didn't look at the different repository.

closed 2024-08-19 by rei-vilo