arduino/ArduinoCore-mbed #942: Nano RP2040 - Standard pin names unknown

rei-vilo, 2024-08-20:
Standard pin names as `D10` are unknown for the Nano RP2040.

- Arduino IDE 2.3.2
- Arduino Mbed OS Nano Boards 4.1.5 
- FQBN: `FQBN: arduino:mbed_nano:nanorp2040connect` 

```
loading library from /home/reivilo/Projets/Arduino/libraries/Adafruit_GC9A01A: invalid library: no header files found
FQBN: arduino:mbed_nano:nanorp2040connect
Using board 'nanorp2040connect' from platform in folder: /home/reivilo/.arduino15/packages/arduino/hardware/mbed_nano/4.1.5
Using core 'arduino' from platform in folder: /home/reivilo/.arduino15/packages/arduino/hardware/mbed_nano/4.1.5

Detecting libraries used...
/home/reivilo/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -w -g3 -nostdlib @/home/reivilo/.arduino15/packages/arduino/hardware/mbed_nano/4.1.5/variants/NANO_RP2040_CONNECT/defines.txt @/home/reivilo/.arduino15/packages/arduino/hardware/mbed_nano/4.1.5/variants/NANO_RP2040_CONNECT/cxxflags.txt -DARDUINO_ARCH_RP2040 -mcpu=cortex-m0plus -w -x c++ -E -CC -DARDUINO=10607 -DARDUINO_NANO_RP2040_CONNECT -DARDUINO_ARCH_MBED_NANO -DARDUINO_ARCH_MBED -DARDUINO_LIBRARY_DISCOVERY_PHASE=1 -I/home/reivilo/.arduino15/packages/arduino/hardware/mbed_nano/4.1.5/cores/arduino -I/home/reivilo/.arduino15/packages/arduino/hardware/mbed_nano/4.1.5/variants/NANO_RP2040_CONNECT -I/home/reivilo/.arduino15/packages/arduino/hardware/mbed_nano/4.1.5/cores/arduino/api/deprecated -I/home/reivilo/.arduino15/packages/arduino/hardware/mbed_nano/4.1.5/cores/arduino/api/deprecated-avr-comp -iprefix/home/reivilo/.arduino15/packages/arduino/hardware/mbed_nano/4.1.5/cores/arduino @/home/reivilo/.arduino15/packages/arduino/hardware/mbed_nano/4.1.5/variants/NANO_RP2040_CONNECT/includes.txt /home/reivilo/.var/app/cc.arduino.IDE2/cache/arduino/sketches/89BF0343D13F509FEFB613C38497C176/sketch/Arduino_Nano_Pins.ino.cpp -o /dev/null
Generating function prototypes...
/home/reivilo/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -w -g3 -nostdlib @/home/reivilo/.arduino15/packages/arduino/hardware/mbed_nano/4.1.5/variants/NANO_RP2040_CONNECT/defines.txt @/home/reivilo/.arduino15/packages/arduino/hardware/mbed_nano/4.1.5/variants/NANO_RP2040_CONNECT/cxxflags.txt -DARDUINO_ARCH_RP2040 -mcpu=cortex-m0plus -w -x c++ -E -CC -DARDUINO=10607 -DARDUINO_NANO_RP2040_CONNECT -DARDUINO_ARCH_MBED_NANO -DARDUINO_ARCH_MBED -DARDUINO_LIBRARY_DISCOVERY_PHASE=1 -I/home/reivilo/.arduino15/packages/arduino/hardware/mbed_nano/4.1.5/cores/arduino -I/home/reivilo/.arduino15/packages/arduino/hardware/mbed_nano/4.1.5/variants/NANO_RP2040_CONNECT -I/home/reivilo/.arduino15/packages/arduino/hardware/mbed_nano/4.1.5/cores/arduino/api/deprecated -I/home/reivilo/.arduino15/packages/arduino/hardware/mbed_nano/4.1.5/cores/arduino/api/deprecated-avr-comp -iprefix/home/reivilo/.arduino15/packages/arduino/hardware/mbed_nano/4.1.5/cores/arduino @/home/reivilo/.arduino15/packages/arduino/hardware/mbed_nano/4.1.5/variants/NANO_RP2040_CONNECT/includes.txt /home/reivilo/.var/app/cc.arduino.IDE2/cache/arduino/sketches/89BF0343D13F509FEFB613C38497C176/sketch/Arduino_Nano_Pins.ino.cpp -o /home/reivilo/.var/app/cc.arduino.IDE2/cache/450851352/sketch_merged.cpp
/home/reivilo/.arduino15/packages/builtin/tools/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /home/reivilo/.var/app/cc.arduino.IDE2/cache/450851352/sketch_merged.cpp
Compiling sketch...
/home/reivilo/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -w -g3 -nostdlib @/home/reivilo/.arduino15/packages/arduino/hardware/mbed_nano/4.1.5/variants/NANO_RP2040_CONNECT/defines.txt @/home/reivilo/.arduino15/packages/arduino/hardware/mbed_nano/4.1.5/variants/NANO_RP2040_CONNECT/cxxflags.txt -DARDUINO_ARCH_RP2040 -MMD -mcpu=cortex-m0plus -DARDUINO=10607 -DARDUINO_NANO_RP2040_CONNECT -DARDUINO_ARCH_MBED_NANO -DARDUINO_ARCH_MBED -DARDUINO_LIBRARY_DISCOVERY_PHASE=0 -I/home/reivilo/.arduino15/packages/arduino/hardware/mbed_nano/4.1.5/cores/arduino -I/home/reivilo/.arduino15/packages/arduino/hardware/mbed_nano/4.1.5/variants/NANO_RP2040_CONNECT -I/home/reivilo/.arduino15/packages/arduino/hardware/mbed_nano/4.1.5/cores/arduino/api/deprecated -I/home/reivilo/.arduino15/packages/arduino/hardware/mbed_nano/4.1.5/cores/arduino/api/deprecated-avr-comp -iprefix/home/reivilo/.arduino15/packages/arduino/hardware/mbed_nano/4.1.5/cores/arduino @/home/reivilo/.arduino15/packages/arduino/hardware/mbed_nano/4.1.5/variants/NANO_RP2040_CONNECT/includes.txt /home/reivilo/.var/app/cc.arduino.IDE2/cache/arduino/sketches/89BF0343D13F509FEFB613C38497C176/sketch/Arduino_Nano_Pins.ino.cpp -o /home/reivilo/.var/app/cc.arduino.IDE2/cache/arduino/sketches/89BF0343D13F509FEFB613C38497C176/sketch/Arduino_Nano_Pins.ino.cpp.o
/home/reivilo/Projets/Arduino/Arduino_Nano_Pins/Arduino_Nano_Pins.ino:49:16: error: 'D10' was not declared in this scope
   .panelBusy = D10,  ///< EXT3 and EXT3-1 pin 3 Red -> D10
                ^~~
/home/reivilo/Projets/Arduino/Arduino_Nano_Pins/Arduino_Nano_Pins.ino:49:16: note: suggested alternative: 'B10'
   .panelBusy = D10,  ///< EXT3 and EXT3-1 pin 3 Red -> D10
                ^~~
                B10
/home/reivilo/Projets/Arduino/Arduino_Nano_Pins/Arduino_Nano_Pins.ino:50:14: error: 'D9' was not declared in this scope
   .panelDC = D9,     ///< EXT3 and EXT3-1 pin 4 Orange -> D9
              ^~
/home/reivilo/Projets/Arduino/Arduino_Nano_Pins/Arduino_Nano_Pins.ino:50:14: note: suggested alternative: 'p9'
   .panelDC = D9,     ///< EXT3 and EXT3-1 pin 4 Orange -> D9
              ^~
              p9
/home/reivilo/Projets/Arduino/Arduino_Nano_Pins/Arduino_Nano_Pins.ino:51:17: error: 'D8' was not declared in this scope
   .panelReset = D8,  ///< EXT3 and EXT3-1 pin 5 Yellow -> D8
                 ^~
/home/reivilo/Projets/Arduino/Arduino_Nano_Pins/Arduino_Nano_Pins.ino:51:17: note: suggested alternative: 'p8'
   .panelReset = D8,  ///< EXT3 and EXT3-1 pin 5 Yellow -> D8
                 ^~
                 p8
/home/reivilo/Projets/Arduino/Arduino_Nano_Pins/Arduino_Nano_Pins.ino:52:14: error: 'D7' was not declared in this scope
   .flashCS = D7,     ///< EXT3 and EXT3-1 pin 8 Violet -> D7
              ^~
/home/reivilo/Projets/Arduino/Arduino_Nano_Pins/Arduino_Nano_Pins.ino:52:14: note: suggested alternative: 'A7'
   .flashCS = D7,     ///< EXT3 and EXT3-1 pin 8 Violet -> D7
              ^~
              A7
/home/reivilo/Projets/Arduino/Arduino_Nano_Pins/Arduino_Nano_Pins.ino:53:14: error: 'D6' was not declared in this scope
   .panelCS = D6,     ///< EXT3 and EXT3-1 pin 9 Grey -> D6
              ^~
/home/reivilo/Projets/Arduino/Arduino_Nano_Pins/Arduino_Nano_Pins.ino:53:14: note: suggested alternative: 'A6'
   .panelCS = D6,     ///< EXT3 and EXT3-1 pin 9 Grey -> D6
              ^~
              A6
/home/reivilo/Projets/Arduino/Arduino_Nano_Pins/Arduino_Nano_Pins.ino:56:15: error: 'D5' was not declared in this scope
   .flashCSS = D5,               ///< EXT3 pin 20 or EXT3-1 pin 11 Black2 -> D5
               ^~
/home/reivilo/Projets/Arduino/Arduino_Nano_Pins/Arduino_Nano_Pins.ino:56:15: note: suggested alternative: 'A5'
   .flashCSS = D5,               ///< EXT3 pin 20 or EXT3-1 pin 11 Black2 -> D5
               ^~
               A5
/home/reivilo/Projets/Arduino/Arduino_Nano_Pins/Arduino_Nano_Pins.ino:59:17: error: 'D2' was not declared in this scope
   .panelPower = D2,             ///< EXT4 pin 20 White -> D2
                 ^~
/home/reivilo/Projets/Arduino/Arduino_Nano_Pins/Arduino_Nano_Pins.ino:59:17: note: suggested alternative: 'A2'
   .panelPower = D2,             ///< EXT4 pin 20 White -> D2
                 ^~
                 A2
/home/reivilo/Projets/Arduino/Arduino_Nano_Pins/Arduino_Nano_Pins.ino:64:13: error: 'D3' was not declared in this scope
   .button = D3,   // EXT4 pin 12 Grey -> D3
             ^~
/home/reivilo/Projets/Arduino/Arduino_Nano_Pins/Arduino_Nano_Pins.ino:64:13: note: suggested alternative: 'A3'
   .button = D3,   // EXT4 pin 12 Grey -> D3
             ^~
             A3
/home/reivilo/Projets/Arduino/Arduino_Nano_Pins/Arduino_Nano_Pins.ino:65:14: error: 'D4' was not declared in this scope
   .ledData = D4,  // EXT4 pin 13 Violet WS2813C -> D4
              ^~
/home/reivilo/Projets/Arduino/Arduino_Nano_Pins/Arduino_Nano_Pins.ino:65:14: note: suggested alternative: 'A4'
   .ledData = D4,  // EXT4 pin 13 Violet WS2813C -> D4
              ^~
              A4

exit status 1

Compilation error: 'D10' was not declared in this scope
```

JAndrassy, 2024-08-20:
why do you think these should be defined? 
use pin numbers

rei-vilo, 2024-08-21:
Because the `D4` is defined for the other Nano boards: ESP32, BLE, Matter.

JAndrassy, 2024-08-21:
but not for Uno, classic Nano, Mega, Zero
ESP32 and Matter cores are not by Arduino