arduino/ArduinoCore-mbed #944: GIGA R1, WiFiUDP::endPacket() fails after release 4.1.3

giorgiogallina, 2024-08-22:
Dear developers,

following issue #858, which is still making me struggle, I am now facing the following issue:
If I try to send an UDP packet in Access Point mode, opened after having unsuccesfully tried to connect to some networks, the function `endPacket()` always fails after updating to 4.1.3 release of Arduino Mbed OS GIGA Boards. The problem persists in release 4.1.5.

A bit of background: in my project, I need to:
1. try to connect to some WiFi network,
2. If/when the connection fails, open Access Point and, through a captive portal, ask user to insert SSID and pwd to their WiFi connection (needed on first start and later, in case of credentials changes, for instance),
3. Connect to the WiFi with the new credentials.

I am attaching a simplified version of my code below, but I guess you can easily build up one on your own.

Many thanks for your contribution!

```
/*
  Adaptation of examples created by Tom Igoe and modified by Karl Söderby
 */

#include <WiFi.h>
#include <WiFiUdp.h>
char ssid[] = "arduinoAP";  // your network SSID (name)
char pass[] = "qwerty123";  // your network password (use for WPA, or use as key for WEP)

#define DBGON  // Debug option  -serial print
#define DBGONP
#define DBGON_X   // Debug option - incl UDP (DNS) packets

// Define UDP settings for DNS 
#define UDP_PACKET_SIZE 1024          // UDP packet size time out, preventign too large packet reads
#define DNS_HEADER_SIZE 12     // DNS Header
#define DNS_ANSWER_SIZE 16     // DNS Answer = standard set with Packet Compression
#define DNS_MAX_REQUESTS 32    // trigger first DNS requests, to redirect to own web-page
#define AP_DNS_PORT  53        // local port to listen for UDP packets
#define AP_CHANNEL  5          // AP wifi channel

WiFiUDP udp;       // A UDP instance to let us send and receive packets over UDP
IPAddress ap_ip;          // Global Acces Point IP adress

int status = WL_IDLE_STATUS;
WiFiServer server(80);

bool AP_DNSRedirect(void)
{
  int t = 0;     // loop index
  int flag = 0;  // conditions check
  uint16_t packet_size = 0;
  uint16_t reply_size = 0;
  IPAddress client_ip;
  int client_port;
  byte packet_buffer[UDP_PACKET_SIZE + DNS_ANSWER_SIZE];   // buffer to hold incoming and outgoing packets

  // get packet size, if available
  packet_size = udp.parsePacket();

  // something went wrong, return with error
  if (packet_size > UDP_PACKET_SIZE)
  {
    Serial.println("DNS INFO: too large packet Error, size " + String(packet_size));
    return false;
  }

  // packet available, read it
  if (packet_size > 0)
  {
    // read packet from client
    int plen = udp.read(packet_buffer, packet_size);
    if (plen > 0  &&  plen < UDP_PACKET_SIZE)   // just to be sure
    { packet_buffer[plen] = 0;
    }
[... 224 lines cut ...]
          digitalWrite(LED_RED, LOW);  
        }
        if (currentLine.endsWith("GET /LR")) {
          digitalWrite(LED_RED, HIGH); 
        }
        if (currentLine.endsWith("GET /HG")) {
          digitalWrite(LED_GREEN, LOW); 
        }
        if (currentLine.endsWith("GET /LG")) {
          digitalWrite(LED_GREEN, HIGH);  
        }
        if (currentLine.endsWith("GET /BH")) {
          digitalWrite(LED_BLUE, LOW);  
        }
        if (currentLine.endsWith("GET /BL")) {
          digitalWrite(LED_BLUE, HIGH);  
        }
      }
    }
    // close the connection:
    client.stop();
    Serial.println("client disconnected");
  }
}
```

giorgiogallina, 2024-09-06:
Some further details that might help.

I was debugging my code and discovered that even on MBED OS Release 4.1.1 WiFiUDP::endPacket() fails if it was preceded by some attempts to connect with a *wrong password* to an exisiting network. If I try to connect to some networks that does not exist, on the other hand, everything works fine.

I was tempted to open a new issue and link it here, but I guess it can be treated as a single issue.

Thank you

maidnl, 2024-09-30:
@giorgiogallina  thanks for the detailed report. I submitted a PR that should solve the problem. However I have just little doubt: your code is meant to fail the WiFi connection to a non existent WiFi Network, but what if the Network exists? I suppose the AP is no longer created... Am I right?
If fail create the AP, if not use the WiFi connection as station.

giorgiogallina, 2024-10-01:
Thank you @maidnl for your work. I will try it and let you know

Yes you are correct about our code.
The AP is needed to get the SSID and password to be used for connection in STA mode. Then, any time the connection fails a few times in a row, we open the AP mode again, in case the user has changed network credentials. If nowone connects for 10 minutes, we try connecting in STA again (maybe there was not a cred. change, just a blackout or whatever), and again open AP on STA failure, iteratively.

giorgiogallina, 2024-10-01:
It works, thank you very much!

closed 2024-10-01 by giorgiogallina