arduino/ArduinoCore-mbed #951: linker_script.ld: No such file or directory

SoothingMist, 2024-09-13:
What I will describe seems to be very similar to https://github.com/arduino/ArduinoCore-mbed/issues/46 but I am not familiar beyond the IDE user level and so cannot understand most of it.

- Using Arduino IDE v2.3.3.

- The board I am working with uses a BSP based on ArduinoCore-mbed.

- Problem occurs upon establishing blank sketch and pressing Verify button.

- Error Message: arm-none-eabi-g++: error: Raeth\AppData\Local\Temp\arduino\sketches\E034F41F9E84F40C81531BEB414A0421/linker_script.ld: No such file or directory

- At issue, I think, is that the user name is “Peter Raeth”. The user directory is C:\Users\Peter Raeth.

- Problem does not occur with the most recent IDE v1.x.

Is there an interim local fix I can apply? If so, exactly how?

Many thanks for your insights and hard work concerning this IDE.

facchinm, 2024-09-16:
Hi @SoothingMist ,
can you add the full compilation output and the core version which you are using?
The issue you are reporting was fixed more than 3 years ago (https://github.com/arduino/ArduinoCore-mbed/blame/main/platform.txt#L85).

SoothingMist, 2024-09-16:
Hello @facchinm. Thank you for your response.

Added to my ticket with RAKwireless to inquire about the mbed version they are using. (Do not know how to find out from the user level.)

Modified IDE preferences so that all compilation output is delivered.

The code is attached, as is the full compilation  output.

Right near the bottom of the compilation messages is:
"arm-none-eabi-g++: error: Raeth\AppData\Local\Temp\arduino\sketches\A5AE74E96AAD4B4B4FFB573F55CA783D/linker_script.ld: No such file or directory"

[CompileErrors.txt](https://github.com/user-attachments/files/17013585/CompileErrors.txt)

[RAK11300_Battery_Level_Detect.ino.txt](https://github.com/user-attachments/files/17013477/RAK11300_Battery_Level_Detect.ino.txt)

facchinm, 2024-09-16:
Got it, they probably never updated their core to match mainline. Closing this one since it doesn't affect our core.
To temporarily fix on your side, you can open `C:\\Users\Peter Raeth\AppData\Local\Arduino15\packages\rakwireless\hardware\mbed_rp2040\0.0.6\platform.txt`
and add the quotes like in [this commit](https://github.com/arduino/ArduinoCore-mbed/commit/07749a46ff2bfd2a7cfd42d7c89378e3c1897fcc#diff-5c378d0844f0422d4d813eabe83f61ffce82014d8b8aa3e93ee35ce2ca14ca7bR85)

closed 2024-09-16 by facchinm

SoothingMist, 2024-09-18:
OH Man Martino.... Thanks much.  Works just fine now. Will pass this result on to RAKwireless. Going to add acknowledgement to my project's documentation.