arduino/ArduinoCore-mbed #955: Ethernet: High ping on Portenta H7 compared to MKR Shield

razvanphp, 2024-09-20:
After migrating our app from MKR WiFi 1010 + ETH shield --> Portenta H7 + Vision Shield, we noticed increased and uneven/unstable ping response times.

Here it is how it looks like on MKR:

```text
64 bytes from 172.30.0.2: icmp_seq=1389 ttl=128 time=0.465 ms
64 bytes from 172.30.0.2: icmp_seq=1390 ttl=128 time=0.315 ms
64 bytes from 172.30.0.2: icmp_seq=1391 ttl=128 time=0.503 ms
64 bytes from 172.30.0.2: icmp_seq=1392 ttl=128 time=0.279 ms
64 bytes from 172.30.0.2: icmp_seq=1393 ttl=128 time=0.657 ms
64 bytes from 172.30.0.2: icmp_seq=1394 ttl=128 time=0.403 ms
64 bytes from 172.30.0.2: icmp_seq=1395 ttl=128 time=0.374 ms
64 bytes from 172.30.0.2: icmp_seq=1396 ttl=128 time=0.304 ms
64 bytes from 172.30.0.2: icmp_seq=1397 ttl=128 time=0.430 ms
64 bytes from 172.30.0.2: icmp_seq=1398 ttl=128 time=0.474 ms
64 bytes from 172.30.0.2: icmp_seq=1399 ttl=128 time=0.463 ms
64 bytes from 172.30.0.2: icmp_seq=1400 ttl=128 time=0.386 ms
64 bytes from 172.30.0.2: icmp_seq=1401 ttl=128 time=0.398 ms
64 bytes from 172.30.0.2: icmp_seq=1402 ttl=128 time=0.399 ms
64 bytes from 172.30.0.2: icmp_seq=1403 ttl=128 time=0.322 ms
64 bytes from 172.30.0.2: icmp_seq=1404 ttl=128 time=0.464 ms
64 bytes from 172.30.0.2: icmp_seq=1405 ttl=128 time=0.394 ms
64 bytes from 172.30.0.2: icmp_seq=1406 ttl=128 time=0.417 ms
64 bytes from 172.30.0.2: icmp_seq=1407 ttl=128 time=0.609 ms
64 bytes from 172.30.0.2: icmp_seq=1408 ttl=128 time=0.429 ms
64 bytes from 172.30.0.2: icmp_seq=1409 ttl=128 time=0.534 ms
64 bytes from 172.30.0.2: icmp_seq=1410 ttl=128 time=0.382 ms
64 bytes from 172.30.0.2: icmp_seq=1411 ttl=128 time=0.621 ms
64 bytes from 172.30.0.2: icmp_seq=1412 ttl=128 time=0.339 ms
64 bytes from 172.30.0.2: icmp_seq=1413 ttl=128 time=0.638 ms
64 bytes from 172.30.0.2: icmp_seq=1414 ttl=128 time=0.335 ms
64 bytes from 172.30.0.2: icmp_seq=1415 ttl=128 time=0.410 ms
64 bytes from 172.30.0.2: icmp_seq=1416 ttl=128 time=0.217 ms
64 bytes from 172.30.0.2: icmp_seq=1417 ttl=128 time=0.463 ms
64 bytes from 172.30.0.2: icmp_seq=1418 ttl=128 time=0.372 ms
64 bytes from 172.30.0.2: icmp_seq=1419 ttl=128 time=0.391 ms
64 bytes from 172.30.0.2: icmp_seq=1420 ttl=128 time=0.603 ms
64 bytes from 172.30.0.2: icmp_seq=1421 ttl=128 time=0.300 ms
64 bytes from 172.30.0.2: icmp_seq=1422 ttl=128 time=0.512 ms
64 bytes from 172.30.0.2: icmp_seq=1423 ttl=128 time=0.378 ms
64 bytes from 172.30.0.2: icmp_seq=1424 ttl=128 time=0.509 ms
64 bytes from 172.30.0.2: icmp_seq=1425 ttl=128 time=0.473 ms
64 bytes from 172.30.0.2: icmp_seq=1426 ttl=128 time=0.345 ms
64 bytes from 172.30.0.2: icmp_seq=1427 ttl=128 time=0.344 ms
```

And here it is on Portenta H7:

```text
64 bytes from 172.30.0.3: icmp_seq=1 ttl=255 time=7.47 ms
64 bytes from 172.30.0.3: icmp_seq=2 ttl=255 time=2.37 ms
64 bytes from 172.30.0.3: icmp_seq=3 ttl=255 time=1.69 ms
64 bytes from 172.30.0.3: icmp_seq=4 ttl=255 time=4.95 ms
64 bytes from 172.30.0.3: icmp_seq=5 ttl=255 time=4.92 ms
64 bytes from 172.30.0.3: icmp_seq=6 ttl=255 time=4.93 ms
64 bytes from 172.30.0.3: icmp_seq=7 ttl=255 time=0.404 ms
64 bytes from 172.30.0.3: icmp_seq=8 ttl=255 time=1.74 ms
64 bytes from 172.30.0.3: icmp_seq=9 ttl=255 time=0.145 ms
64 bytes from 172.30.0.3: icmp_seq=10 ttl=255 time=1.75 ms
64 bytes from 172.30.0.3: icmp_seq=11 ttl=255 time=0.710 ms
64 bytes from 172.30.0.3: icmp_seq=12 ttl=255 time=1.76 ms
64 bytes from 172.30.0.3: icmp_seq=13 ttl=255 time=0.686 ms
64 bytes from 172.30.0.3: icmp_seq=14 ttl=255 time=1.75 ms
64 bytes from 172.30.0.3: icmp_seq=15 ttl=255 time=0.186 ms
64 bytes from 172.30.0.3: icmp_seq=16 ttl=255 time=1.77 ms
64 bytes from 172.30.0.3: icmp_seq=17 ttl=255 time=0.583 ms
64 bytes from 172.30.0.3: icmp_seq=18 ttl=255 time=1.82 ms
64 bytes from 172.30.0.3: icmp_seq=19 ttl=255 time=4.92 ms
64 bytes from 172.30.0.3: icmp_seq=20 ttl=255 time=2.94 ms
64 bytes from 172.30.0.3: icmp_seq=21 ttl=255 time=1.84 ms
64 bytes from 172.30.0.3: icmp_seq=22 ttl=255 time=0.561 ms
64 bytes from 172.30.0.3: icmp_seq=23 ttl=255 time=1.89 ms
64 bytes from 172.30.0.3: icmp_seq=24 ttl=255 time=0.788 ms
64 bytes from 172.30.0.3: icmp_seq=25 ttl=255 time=1.89 ms
64 bytes from 172.30.0.3: icmp_seq=26 ttl=255 time=4.95 ms
64 bytes from 172.30.0.3: icmp_seq=27 ttl=255 time=4.07 ms
64 bytes from 172.30.0.3: icmp_seq=28 ttl=255 time=3.96 ms
64 bytes from 172.30.0.3: icmp_seq=29 ttl=255 time=2.94 ms
64 bytes from 172.30.0.3: icmp_seq=30 ttl=255 time=1.65 ms
64 bytes from 172.30.0.3: icmp_seq=31 ttl=255 time=4.93 ms
```

I suspect this is impacting our app's performance, since we need sub ms processing of the data received over ethernet, but not sure how to approach this problem.

Userland code is basically inexistent, as I'm testing just with this:

```cpp
#include <PortentaEthernet.h>
#include <Ethernet.h>

EthernetClient ethClient;

byte mac[] = { 0xA8, 0x61, 0x0A, 0xAE, 0x44, 0xE0 };
IPAddress ip(172, 30, 0, 3);

void setup() {
  Ethernet.begin(mac, ip);
}

void loop() {}
```

Initially I though this is because MKR ETH Shield uses W5500 chipset with a built-in hardware TCP/IP stack, while Portenta H7 uses LAN8742AI, an Ethernet PHY that requires a software TCP/IP stack like LwIP.

But today I tried to flash the exact same board with micropython and ping times look perfect:
```
64 bytes from 192.168.10.252: icmp_seq=10 ttl=255 time=0.075 ms
64 bytes from 192.168.10.252: icmp_seq=11 ttl=255 time=0.086 ms
64 bytes from 192.168.10.252: icmp_seq=12 ttl=255 time=0.082 ms
64 bytes from 192.168.10.252: icmp_seq=13 ttl=255 time=0.069 ms
64 bytes from 192.168.10.252: icmp_seq=14 ttl=255 time=0.072 ms
64 bytes from 192.168.10.252: icmp_seq=15 ttl=255 time=0.092 ms
64 bytes from 192.168.10.252: icmp_seq=16 ttl=255 time=0.094 ms
64 bytes from 192.168.10.252: icmp_seq=17 ttl=255 time=0.095 ms
64 bytes from 192.168.10.252: icmp_seq=18 ttl=255 time=0.085 ms
64 bytes from 192.168.10.252: icmp_seq=19 ttl=255 time=0.067 ms
64 bytes from 192.168.10.252: icmp_seq=20 ttl=255 time=0.094 ms
64 bytes from 192.168.10.252: icmp_seq=21 ttl=255 time=0.071 ms
64 bytes from 192.168.10.252: icmp_seq=22 ttl=255 time=0.086 ms
64 bytes from 192.168.10.252: icmp_seq=23 ttl=255 time=0.084 ms
64 bytes from 192.168.10.252: icmp_seq=24 ttl=255 time=0.069 ms
64 bytes from 192.168.10.252: icmp_seq=25 ttl=255 time=0.105 ms
64 bytes from 192.168.10.252: icmp_seq=26 ttl=255 time=0.084 ms
64 bytes from 192.168.10.252: icmp_seq=27 ttl=255 time=0.095 ms
```

Are there any settings or optimisations that we could try in LwIP? Seems the issue is not in HW.

Thank you!
R

PS: opened initially in https://github.com/arduino-libraries/Ethernet/issues/275

razvanphp, 2024-11-07:
anybody any idea?

I tried increasing some values in `mbed_config.h`, similar to micropython ones, ping improved a little bit, but still has spikes of 2-5 ms.

```
#define MBED_CONF_LWIP_ADDR_TIMEOUT                                             5                                                                                                // set by library:lwip
#define MBED_CONF_LWIP_ADDR_TIMEOUT_MODE                                        1                                                                                                // set by library:lwip
#define MBED_CONF_LWIP_DEBUG_ENABLED                                            0                                                                                                // set by library:lwip
#define MBED_CONF_LWIP_DEFAULT_THREAD_STACKSIZE                                 512                                                                                              // set by library:lwip
#define MBED_CONF_LWIP_DHCP_TIMEOUT                                             60                                                                                               // set by library:lwip
#define MBED_CONF_LWIP_ENABLE_PPP_TRACE                                         0                                                                                                // set by library:lwip
#define MBED_CONF_LWIP_ETHERNET_ENABLED                                         1                                                                                                // set by library:lwip
#define MBED_CONF_LWIP_IPV4_ENABLED                                             1                                                                                                // set by library:lwip
#define MBED_CONF_LWIP_IPV6_ENABLED                                             0                                                                                                // set by library:lwip
#define MBED_CONF_LWIP_IP_VER_PREF                                              4                                                                                                // set by library:lwip
#define MBED_CONF_LWIP_L3IP_ENABLED                                             0                                                                                                // set by library:lwip
#define MBED_CONF_LWIP_MBOX_SIZE                                                8                                                                                                // set by library:lwip
#define MBED_CONF_LWIP_MEMP_NUM_TCPIP_MSG_INPKT                                 8                                                                                                // set by library:lwip
#define MBED_CONF_LWIP_MEMP_NUM_TCP_SEG                                         32                                                                                               // set by library:lwip
#define MBED_CONF_LWIP_MEM_SIZE                                                 16000                                                                                            // set by library:lwip[PORTENTA_H7]
#define MBED_CONF_LWIP_ND6_QUEUEING                                             0                                                                                                // set by library:lwip
#define MBED_CONF_LWIP_ND6_RDNSS_MAX_DNS_SERVERS                                0                                                                                                // set by library:lwip
#define MBED_CONF_LWIP_NETBUF_RECVINFO_ENABLED                                  0                                                                                                // set by library:lwip
#define MBED_CONF_LWIP_NUM_NETBUF                                               8                                                                                                // set by library:lwip
#define MBED_CONF_LWIP_NUM_PBUF                                                 8                                                                                                // set by library:lwip
#define MBED_CONF_LWIP_PBUF_POOL_SIZE                                           16                                                                                               // set by library:lwip
#define MBED_CONF_LWIP_PPP_ENABLED                                              0                                                                                                // set by library:lwip
#define MBED_CONF_LWIP_PPP_IPV4_ENABLED                                         0                                                                                                // set by library:lwip
#define MBED_CONF_LWIP_PPP_IPV6_ENABLED                                         0                                                                                                // set by library:lwip
#define MBED_CONF_LWIP_PPP_THREAD_STACKSIZE                                     768                                                                                              // set by library:lwip
#define MBED_CONF_LWIP_PRESENT                                                  1                                                                                                // set by library:lwip
[... cut ...]
#define MBED_CONF_LWIP_TCP_SYNMAXRTX                                            6                                                                                                // set by library:lwip
#define MBED_CONF_LWIP_TCP_WND                                                  (8 * TCP_MSS)                                                                                    // set by library:lwip
#define MBED_CONF_LWIP_UDP_SOCKET_MAX                                           4                                                                                                // set by library:lwip
#define MBED_CONF_LWIP_USE_MBED_TRACE                                           0  
```

pennam, 2024-11-08:
@razvanphp are you getting the same results using WiFi? or WiFi is working good? If it is i would say the root cause is probably at lower level than LwIP.

multiplemonomials, 2025-03-17:
Hi! I just completed rewriting the entire STM32H7 ethernet driver in Mbed. I think there's a good chance it will solve this issue, as the existing one was held together with the digital equivalent of glue and duct tape.

PR here: https://github.com/mbed-ce/mbed-os/pull/438

razvanphp, 2025-03-17:
Did you managed to get better latency with the new driver? Any idea if I can test your branch before mbed merges it?

Not sure how Arduino layer plays on top of mbed, now that mbed is deprecated officially.

razvanphp, 2025-03-17:
Portenta H7. I could replace some files on my machine and clear the cache, it should load it on a fresh build, if it is compatible with arduino ETH API that is built on-top. Thank you!

multiplemonomials, 2025-03-17:
Yeah if you manually apply my patch, it should work. Also I will work on preparing a version of the mbed ce arduino core for you.

multiplemonomials, 2025-04-02:
Good news! I was able to prepare a Portenta release of the Mbed CE arduino core. Still have more checking to do, but I was able to confirm that it at least boots, flashes code, and can talk over Ethernet. Please give it a try if you have a chance! https://github.com/mbed-ce/ArduinoCore-mbed-ce/releases/tag/v1.2.0

razvanphp, 2025-04-07:
I tried to flash your release today, but the basic demo code from this issue with static IP does not make the interface come up, it does not respond to ping. It seems it's using the correct folder/version for the flash:

```
FQBN: arduino:mbed_ce_test:portenta_h7_m7
Using board 'portenta_h7_m7' from platform in folder: /Users/razvan/Library/Arduino15/packages/arduino/hardware/mbed_ce_test/1.1.0
Using core 'arduino' from platform in folder: /Users/razvan/Library/Arduino15/packages/arduino/hardware/mbed_ce_test/1.1.0

Detecting libraries used...
/Users/razvan/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -w -g3 -nostdlib @/Users/razvan/Library/Arduino15/packages/arduino/hardware/mbed_ce_test/1.1.0/variants/ARDUINO_PORTENTA_H7_M7/defines.txt @/Users/razvan/Library/Arduino15/packages/arduino/hardware/mbed_ce_test/1.1.0/variants/ARDUINO_PORTENTA_H7_M7/cxxflags.txt -w -x c++ -E -CC -DARDUINO=10607 -DARDUINO_ARDUINO_PORTENTA_H7_M7 -DARDUINO_ARCH_MBED_CE_TEST -DARDUINO_LIBRARY_DISCOVERY_PHASE=1 -I/Users/razvan/Library/Arduino15/packages/arduino/hardware/mbed_ce_test/1.1.0/cores/arduino -I/Users/razvan/Library/Arduino15/packages/arduino/hardware/mbed_ce_test/1.1.0/variants/ARDUINO_PORTENTA_H7_M7 -I/Users/razvan/Library/Arduino15/packages/arduino/hardware/mbed_ce_test/1.1.0/cores/arduino/api/deprecated -I/Users/razvan/Library/Arduino15/packages/arduino/hardware/mbed_ce_test/1.1.0/cores/arduino/api/deprecated-avr-comp -iprefix/Users/razvan/Library/Arduino15/packages/arduino/hardware/mbed_ce_test/1.1.0/cores/arduino @/Users/razvan/Library/Arduino15/packages/arduino/hardware/mbed_ce_test/1.1.0/variants/ARDUINO_PORTENTA_H7_M7/includes.txt /Users/razvan/Library/Caches/arduino/sketches/5F7B38B7ACE97AB5A4ED558FDF30C7B6/sketch/sketch_apr2a.ino.cpp -o /dev/null
Alternatives for Ethernet.h: [Ethernet@2.0.2]
ResolveLibrary(Ethernet.h)
  -> candidates: [Ethernet@2.0.2]
/Users/razvan/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -w -g3 -nostdlib @/Users/razvan/Library/Arduino15/packages/arduino/hardware/mbed_ce_test/1.1.0/variants/ARDUINO_PORTENTA_H7_M7/defines.txt @/Users/razvan/Library/Arduino15/packages/arduino/hardware/mbed_ce_test/1.1.0/variants/ARDUINO_PORTENTA_H7_M7/cxxflags.txt -w -x c++ -E -CC -DARDUINO=10607 -DARDUINO_ARDUINO_PORTENTA_H7_M7 -DARDUINO_ARCH_MBED_CE_TEST -DARDUINO_LIBRARY_DISCOVERY_PHASE=1 -I/Users/razvan/Library/Arduino15/packages/arduino/hardware/mbed_ce_test/1.1.0/cores/arduino -I/Users/razvan/Library/Arduino15/packages/arduino/hardware/mbed_ce_test/1.1.0/variants/ARDUINO_PORTENTA_H7_M7 -I/Users/razvan/Library/Arduino15/libraries/Ethernet/src -I/Users/razvan/Library/Arduino15/packages/arduino/hardware/mbed_ce_test/1.1.0/cores/arduino/api/deprecated -I/Users/razvan/Library/Arduino15/packages/arduino/hardware/mbed_ce_test/1.1.0/cores/arduino/api/deprecated-avr-comp -iprefix/Users/razvan/Library/Arduino15/packages/arduino/hardware/mbed_ce_test/1.1.0/cores/arduino @/Users/razvan/Library/Arduino15/packages/arduino/hardware/mbed_ce_test/1.1.0/variants/ARDUINO_PORTENTA_H7_M7/includes.txt /Users/razvan/Library/Caches/arduino/sketches/5F7B38B7ACE97AB5A4ED558FDF30C7B6/sketch/sketch_apr2a.ino.cpp -o /dev/null
/Users/razvan/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -w -g3 -nostdlib @/Users/razvan/Library/Arduino15/packages/arduino/hardware/mbed_ce_test/1.1.0/variants/ARDUINO_PORTENTA_H7_M7/defines.txt @/Users/razvan/Library/Arduino15/packages/arduino/hardware/mbed_ce_test/1.1.0/variants/ARDUINO_PORTENTA_H7_M7/cxxflags.txt -w -x c++ -E -CC -DARDUINO=10607 -DARDUINO_ARDUINO_PORTENTA_H7_M7 -DARDUINO_ARCH_MBED_CE_TEST -DARDUINO_LIBRARY_DISCOVERY_PHASE=1 -I/Users/razvan/Library/Arduino15/packages/arduino/hardware/mbed_ce_test/1.1.0/cores/arduino -I/Users/razvan/Library/Arduino15/packages/arduino/hardware/mbed_ce_test/1.1.0/variants/ARDUINO_PORTENTA_H7_M7 -I/Users/razvan/Library/Arduino15/libraries/Ethernet/src -I/Users/razvan/Library/Arduino15/packages/arduino/hardware/mbed_ce_test/1.1.0/cores/arduino/api/deprecated -I/Users/razvan/Library/Arduino15/packages/arduino/hardware/mbed_ce_test/1.1.0/cores/arduino/api/deprecated-avr-comp -iprefix/Users/razvan/Library/Arduino15/packages/arduino/hardware/mbed_ce_test/1.1.0/cores/arduino @/Users/razvan/Library/Arduino15/packages/arduino/hardware/mbed_ce_test/1.1.0/variants/ARDUINO_PORTENTA_H7_M7/includes.txt /Users/razvan/Library/Arduino15/libraries/Ethernet/src/Dhcp.cpp -o /dev/null
Alternatives for SPI.h: [SPI]
ResolveLibrary(SPI.h)
  -> candidates: [SPI]
/Users/razvan/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -c -w -g3 -nostdlib @/Users/razvan/Library/Arduino15/packages/arduino/hardware/mbed_ce_test/1.1.0/variants/ARDUINO_PORTENTA_H7_M7/defines.txt @/Users/razvan/Library/Arduino15/packages/arduino/hardware/mbed_ce_test/1.1.0/variants/ARDUINO_PORTENTA_H7_M7/cxxflags.txt -w -x c++ -E -CC -DARDUINO=10607 -DARDUINO_ARDUINO_PORTENTA_H7_M7 -DARDUINO_ARCH_MBED_CE_TEST -DARDUINO_LIBRARY_DISCOVERY_PHASE=1 -I/Users/razvan/Library/Arduino15/packages/arduino/hardware/mbed_ce_test/1.1.0/cores/arduino -I/Users/razvan/Library/Arduino15/packages/arduino/hardware/mbed_ce_test/1.1.0/variants/ARDUINO_PORTENTA_H7_M7 -I/Users/razvan/Library/Arduino15/libraries/Ethernet/src -I/Users/razvan/Library/Arduino15/packages/arduino/hardware/mbed_ce_test/1.1.0/libraries/SPI -I/Users/razvan/Library/Arduino15/ [... cut ...]
Download	[=======================  ]  93%       126976 bytes
Download	[======================== ]  96%       131072 bytes
Download	[=========================] 100%       135576 bytes
Download done.
File downloaded successfully
Transitioning to dfuMANIFEST state
```

... I guess it's using the wrong Ethernet lib/folder. Any ideas?

PS: `PortentaEthernet.h` is gone. I think ArduinoCore is pinned to an even older version of mbed-ce, right? So it would never receive your update....

Thank again for working on this!

multiplemonomials, 2025-04-08:
Ah I see the issue, I had not enabled building the Ethernet library in my version of the core so it was falling back to the default one (which relies on the external chip). That's why PortentaEthernet.h was missing. I just pushed a [new release](https://github.com/mbed-ce/ArduinoCore-mbed-ce/releases/tag/v1.2.1) of the core that has this library. Checked it with your code and was able to ping!
```
Pinging 169.254.52.80 with 32 bytes of data:
Reply from 169.254.52.80: bytes=32 time<1ms TTL=255
Reply from 169.254.52.80: bytes=32 time=5ms TTL=255
Reply from 169.254.52.80: bytes=32 time=1ms TTL=255
Reply from 169.254.52.80: bytes=32 time=1ms TTL=255
Reply from 169.254.52.80: bytes=32 time<1ms TTL=255
Reply from 169.254.52.80: bytes=32 time=1ms TTL=255
Reply from 169.254.52.80: bytes=32 time<1ms TTL=255
Reply from 169.254.52.80: bytes=32 time=4ms TTL=255
Reply from 169.254.52.80: bytes=32 time<1ms TTL=255
Reply from 169.254.52.80: bytes=32 time=1ms TTL=255
Reply from 169.254.52.80: bytes=32 time<1ms TTL=255
Reply from 169.254.52.80: bytes=32 time<1ms TTL=255
Reply from 169.254.52.80: bytes=32 time<1ms TTL=255
Reply from 169.254.52.80: bytes=32 time<1ms TTL=255
Reply from 169.254.52.80: bytes=32 time<1ms TTL=255
Reply from 169.254.52.80: bytes=32 time=1ms TTL=255
Reply from 169.254.52.80: bytes=32 time=1ms TTL=255
Reply from 169.254.52.80: bytes=32 time<1ms TTL=255
Reply from 169.254.52.80: bytes=32 time<1ms TTL=255
Reply from 169.254.52.80: bytes=32 time<1ms TTL=255
Reply from 169.254.52.80: bytes=32 time<1ms TTL=255
Reply from 169.254.52.80: bytes=32 time<1ms TTL=255
Reply from 169.254.52.80: bytes=32 time<1ms TTL=255
Reply from 169.254.52.80: bytes=32 time<1ms TTL=255
Reply from 169.254.52.80: bytes=32 time<1ms TTL=255
Reply from 169.254.52.80: bytes=32 time=2ms TTL=255
Reply from 169.254.52.80: bytes=32 time<1ms TTL=255
Reply from 169.254.52.80: bytes=32 time<1ms TTL=255
Reply from 169.254.52.80: bytes=32 time<1ms TTL=255
Reply from 169.254.52.80: bytes=32 time<1ms TTL=255
Reply from 169.254.52.80: bytes=32 time<1ms TTL=255
Reply from 169.254.52.80: bytes=32 time<1ms TTL=255
Reply from 169.254.52.80: bytes=32 time<1ms TTL=255
Reply from 169.254.52.80: bytes=32 time=1ms TTL=255
Reply from 169.254.52.80: bytes=32 time<1ms TTL=255
Reply from 169.254.52.80: bytes=32 time<1ms TTL=255
Reply from 169.254.52.80: bytes=32 time=1ms TTL=255
Reply from 169.254.52.80: bytes=32 time=1ms TTL=255
Reply from 169.254.52.80: bytes=32 time=1ms TTL=255
Reply from 169.254.52.80: bytes=32 time=1ms TTL=255

Ping statistics for 169.254.52.80:
    Packets: Sent = 40, Received = 40, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 0ms, Maximum = 5ms, Average = 0ms
Control-C
```

razvanphp, 2025-04-08:
nice, it seems to work way better than the original mbed implementation!

Still, I can see some spikes in reponse time, which are very strange, given that the sketch is empty and nothing else is being processed than Ethernet packets.

Would it be possible to still tune the LWIP stack params / buffers for better performance?

```
PING 192.168.10.123 (192.168.10.123) 56(84) bytes of data.
64 bytes from 192.168.10.123: icmp_seq=1 ttl=255 time=7.50 ms
64 bytes from 192.168.10.123: icmp_seq=2 ttl=255 time=0.705 ms
64 bytes from 192.168.10.123: icmp_seq=3 ttl=255 time=0.438 ms
64 bytes from 192.168.10.123: icmp_seq=4 ttl=255 time=0.453 ms
64 bytes from 192.168.10.123: icmp_seq=5 ttl=255 time=0.448 ms
64 bytes from 192.168.10.123: icmp_seq=6 ttl=255 time=0.449 ms
64 bytes from 192.168.10.123: icmp_seq=7 ttl=255 time=0.504 ms
64 bytes from 192.168.10.123: icmp_seq=8 ttl=255 time=3.48 ms
64 bytes from 192.168.10.123: icmp_seq=9 ttl=255 time=0.770 ms
64 bytes from 192.168.10.123: icmp_seq=10 ttl=255 time=0.524 ms
64 bytes from 192.168.10.123: icmp_seq=11 ttl=255 time=0.545 ms
64 bytes from 192.168.10.123: icmp_seq=12 ttl=255 time=0.472 ms
64 bytes from 192.168.10.123: icmp_seq=13 ttl=255 time=0.489 ms
64 bytes from 192.168.10.123: icmp_seq=14 ttl=255 time=0.602 ms
64 bytes from 192.168.10.123: icmp_seq=15 ttl=255 time=0.517 ms
64 bytes from 192.168.10.123: icmp_seq=16 ttl=255 time=0.524 ms
64 bytes from 192.168.10.123: icmp_seq=17 ttl=255 time=3.58 ms
64 bytes from 192.168.10.123: icmp_seq=18 ttl=255 time=1.77 ms
64 bytes from 192.168.10.123: icmp_seq=19 ttl=255 time=0.799 ms
64 bytes from 192.168.10.123: icmp_seq=20 ttl=255 time=0.511 ms
64 bytes from 192.168.10.123: icmp_seq=21 ttl=255 time=0.521 ms
64 bytes from 192.168.10.123: icmp_seq=22 ttl=255 time=0.408 ms
64 bytes from 192.168.10.123: icmp_seq=23 ttl=255 time=0.422 ms
64 bytes from 192.168.10.123: icmp_seq=24 ttl=255 time=0.478 ms
64 bytes from 192.168.10.123: icmp_seq=25 ttl=255 time=0.486 ms
64 bytes from 192.168.10.123: icmp_seq=26 ttl=255 time=0.413 ms
64 bytes from 192.168.10.123: icmp_seq=27 ttl=255 time=0.488 ms
64 bytes from 192.168.10.123: icmp_seq=28 ttl=255 time=3.53 ms
64 bytes from 192.168.10.123: icmp_seq=29 ttl=255 time=1.70 ms
64 bytes from 192.168.10.123: icmp_seq=30 ttl=255 time=0.855 ms
64 bytes from 192.168.10.123: icmp_seq=31 ttl=255 time=0.487 ms
64 bytes from 192.168.10.123: icmp_seq=32 ttl=255 time=0.565 ms
64 bytes from 192.168.10.123: icmp_seq=33 ttl=255 time=3.58 ms
64 bytes from 192.168.10.123: icmp_seq=34 ttl=255 time=3.75 ms
64 bytes from 192.168.10.123: icmp_seq=35 ttl=255 time=0.751 ms
64 bytes from 192.168.10.123: icmp_seq=36 ttl=255 time=3.55 ms
64 bytes from 192.168.10.123: icmp_seq=37 ttl=255 time=1.74 ms
64 bytes from 192.168.10.123: icmp_seq=38 ttl=255 time=0.881 ms
64 bytes from 192.168.10.123: icmp_seq=39 ttl=255 time=0.555 ms
64 bytes from 192.168.10.123: icmp_seq=40 ttl=255 time=1.51 ms
64 bytes from 192.168.10.123: icmp_seq=41 ttl=255 time=0.641 ms
64 bytes from 192.168.10.123: icmp_seq=42 ttl=255 time=0.571 ms
64 bytes from 192.168.10.123: icmp_seq=43 ttl=255 time=0.491 ms
^C
--- 192.168.10.123 ping statistics ---
43 packets transmitted, 43 received, 0% packet loss, time 704ms
rtt min/avg/max/mdev = 0.408/1.242/7.496/1.434 ms
```

multiplemonomials, 2025-04-08:
I think, to debug this further, you could try adding lines inside Mbed like
```
#include <us_ticker_api.h>
...
printf("Current time as of SOME_EVENT: %" PRIi64 "us", ticker_read_us(get_us_ticker_data()));
```
This would let you print the absolute time where network packet events happen.

Like, the first places you could add that would be [here](https://github.com/mbed-ce/mbed-os/blob/0b6a6d1ebd79b97624ca5fb64ab5058de57924e8/connectivity/drivers/emac/include/GenericEthDMA.h#L639) (for Rx) and [here](https://github.com/mbed-ce/mbed-os/blob/0b6a6d1ebd79b97624ca5fb64ab5058de57924e8/connectivity/drivers/emac/include/GenericEthDMA.h#L306) (for Tx). That would print one timestamp when a packet is received, and another when a packet is sent. If you see that there is a largeish delay between Rx and Tx timestamps, that indicates the delay is likely in the LwIP stack or another higher layer. If you see that there is a small delay between receiving the ping and sending the response, that suggests the Ethernet driver is at fault because the upper layer stack responded quickly to the ping.

This does require recompiling the core, but that is thankfully [way easier](https://github.com/mbed-ce/ArduinoCore-mbed-ce?tab=readme-ov-file#compiling-an-arduino-core-for-mbed-ce) in my version (no monster shell scripts!). You can even build and flash the Ethernet examples directly from CMake, if you want to sidestep the Arduino IDE. (e.g. `ninja flash-Ethernet-example-DhcpAddressPrinter`).

Another option to dig into this would be using something like SEGGER SystemView, if you have a J-Link probe available. That would let you visualize all the context switches involved in receiving the packet and would be a much quicker way to track this down. A while back I made a [plugin](https://github.com/multiplemonomials/MbedSysview) to add it to Mbed, which you could add in to the Arduino core project. It would require a bit of build system hacking though because the CMake files in that repo are set up for my older mbed-cmake project.

razvanphp, 2025-04-16:
Unfortunately I do not have SEGGER  SystemView / J-Link, so I can only test holistically with pings.

Can we first try to add similar values conf values (buffers, memory, etc) with micropython to LWIP stack?

I could not find those in your Arduino core build, what values do they have? defaults? Do I have to recompile mbed to change them as well?

https://github.com/arduino/ArduinoCore-mbed/blob/f7ff4114bc0d89f23b45423e049f5bc17114dcbc/variants/PORTENTA_H7_M7/mbed_config.h#L204-L244

multiplemonomials, 2025-04-16:
Yes, you'd have to recompile Mbed to change those I'm afraid. In Mbed, that config file is [here](https://github.com/mbed-ce/mbed-os/blob/master/connectivity/lwipstack/include/lwipstack/lwipopts.h), and it's currently largely driven by JSON options from [here](https://github.com/mbed-ce/mbed-os/blob/master/connectivity/lwipstack/mbed_lib.json5#L5). You could likely just add all those defines at the end of the file to override the Mbed ones, though not sure if that would cause other issues.

razvanphp, 2025-04-17:
I managed to compile it, but I've got mixed results:,

I had those at the beginning:
```
Current time as of RX_EVENT: 135676004us
Current time as of TX_EVENT: 135680882us
Current time as of RX_EVENT: 135845699us
Current time as of TX_EVENT: 135845892us
Current time as of RX_EVENT: 136433752us
Current time as of TX_EVENT: 136433908us
Current time as of RX_EVENT: 136434117us
Current time as of TX_EVENT: 136527923us
Current time as of RX_EVENT: 136528989us
Current time as of TX_EVENT: 136532906us
Current time as of RX_EVENT: 136847468us
Current time as of TX_EVENT: 136847906us
Current time as of RX_EVENT: 137852580us
Current time as of TX_EVENT: 137852910us
Current time as of RX_EVENT: 137948256us
```

Then I added "\n" at the end of printf and started getting all pings after 9ms:
```
Current time as of RX_EVENT: 68475356us
Current time as of TX_EVENT: 68482929us
Current time as of RX_EVENT: 69480108us
Current time as of TX_EVENT: 69487933us
Current time as of RX_EVENT: 70482261us
Current time as of TX_EVENT: 70489948us
Current time as of RX_EVENT: 71485095us
Current time as of TX_EVENT: 71492941us
Current time as of RX_EVENT: 72489973us
Current time as of TX_EVENT: 72497946us
Current time as of RX_EVENT: 73493626us
Current time as of TX_EVENT: 73501941us
Current time as of RX_EVENT: 74498834us
Current time as of TX_EVENT: 74506955us
```

Then I removed the new line again and ping is 1.6 ms in average, with spikes to 95 ms that can be seen in between those events too:
```
Current time as of RX_EVENT: 38350781us
Current time as of TX_EVENT: 38443979us
Current time as of RX_EVENT: 38539029us
Current time as of TX_EVENT: 38539947us
Current time as of RX_EVENT: 39352021us
Current time as of TX_EVENT: 39352967us
Current time as of RX_EVENT: 40353886us
Current time as of TX_EVENT: 40353943us
Current time as of RX_EVENT: 41355640us
Current time as of TX_EVENT: 41355958us
Current time as of RX_EVENT: 42356990us
Current time as of TX_EVENT: 42357944us
Current time as of RX_EVENT: 43358980us
Current time as of TX_EVENT: 43359959us
Current time as of RX_EVENT: 44360974us
Current time as of TX_EVENT: 44361972us
Current time as of RX_EVENT: 45362864us
Current time as of TX_EVENT: 45362949us
Current time as of RX_EVENT: 46364700us
Current time as of TX_EVENT: 46368960us
Current time as of RX_EVENT: 47366064us
Current time as of TX_EVENT: 47368961us
Current time as of RX_EVENT: 48367627us
Current time as of TX_EVENT: 48368961us
Current time as of RX_EVENT: 49368919us
Current time as of TX_EVENT: 49368963us
Current time as of RX_EVENT: 50370618us
Current time as of TX_EVENT: 50463970us
Current time as of RX_EVENT: 51371780us
Current time as of TX_EVENT: 51371955us
Current time as of RX_EVENT: 52373673us
Current time as of TX_EVENT: 52373970us
Current time as of RX_EVENT: 53375753us
Current time as of TX_EVENT: 53375954us
Current time as of RX_EVENT: 54377672us
Current time as of TX_EVENT: 54377968us
Current time as of RX_EVENT: 55379775us
Current time as of TX_EVENT: 55379954us
Current time as of RX_EVENT: 56381677us
Current time as of TX_EVENT: 56381967us
Current time as of RX_EVENT: 57383779us
Current time as of TX_EVENT: 57383951us
Current time as of RX_EVENT: 58385827us
Current time as of TX_EVENT: 58385966us
Current time as of RX_EVENT: 59387692us
Current time as of TX_EVENT: 59387950us
Current time as of RX_EVENT: 60389780us
Current time as of TX_EVENT: 60389965us
Current time as of RX_EVENT: 61390816us
Current time as of TX_EVENT: 61390972us
Current time as of RX_EVENT: 62392744us
Current time as of TX_EVENT: 62392956us
```

What does this suggest? Can you please also check it locally? Thank you!

multiplemonomials, 2025-04-26:
OK I did some investigation into this issue.  I was able to reproduce it with the Arduino web server example.

I dug into it a bit, and that example program is constantly making nonblocking calls to the network stack in the main loop. Now, with LwIP in the configuration Mbed has it in, how it works is that the internals of the network stack are protected by one mutex. So, either an application thread can be in the network stack internals, or LwIP's own thread can be, but not both. In the default Mbed configuration, LwIP runs at osPriorityNormal, same as the main thread.  I suspect what might be happening is that the main thread is DoS-ing the LwIP stack thread by holding that mutex -- if the main thread has the mutex, then LwIP cannot switch in and reply to the ping.

I don't have hard evidence for this (hard to know for sure without a tool like SEGGER SystemView), but I did find that if I increased LwIP's thread priority to osPriorityHigh, the issue seemed to go away.

So, try recompiling the arduino core with the following option in mbed_app.json's "target_overrides" > "*" section:

```
"lwip.tcpip-thread-priority": "osPriorityHigh"
```

That should produce a measurable decrease in ping times, at least for the examples in this repo.