arduino/ArduinoCore-mbed #970: Portenta CAT.M1/NB IoT GNSS Shield connecting failure

brinata, 2024-10-03:
Hello,

I'm using the Portenta CAT.M1/NB IoT GNSS Shield with the EDGE CONTROL board.

if the application fails the initial connection process because the signal strength is low or because the general latency of the GSM network, the GSM library in not able to reconnect: when I try to run the GSM.begin() again, the application program stop working after receiving the "^SYSSTART" string from the modem:

The board was not able to register to the network...
[000270[INFO][CELL]: AT TX ( 2): AT
[000270[INFO][CELL]: AT TX ( 1): <cr>
[000303[INFO][CELL]: AT RX ( 2): <cr><ln>
[000303[INFO][CELL]: AT RX (11): ^SYSSTART<cr><ln>
[000304[INFO][CELL]: AT RX ( 2): <cr><ln>
[000304[INFO][CELL]: AT RX ( 6): +CIEV:
[000304[INFO][CELL]: AT RX (14):  prov,0,"fallb
[000304[INFO][CELL]: AT RX ( 7): 3gpp"<cr><ln>

The only way to re-start is reset the board.

I'm using the mbed 4.1.15

to simulate the problem I forced the following retry table in the GSM.h file:

const uint16_t _retry_timeout[3] = {1, 2, 4};

Please help me because I can't reset the whole application program to re-start the connectivity ....

[GSMClient.txt](https://github.com/user-attachments/files/17247631/GSMClient.txt)
[sketch_yaml.txt](https://github.com/user-attachments/files/17247639/sketch_yaml.txt)

brinata, 2024-10-28:
I've made some test, with these results:

**moduel non connected yet**
 
 setting the timeout value at 1000 ms and calling the shutdown(), it seems that the library is able to re-try the connection procedure with success

**module already connected**

if you call the GSM.disconnect() function, then the library is not able to recover its initial state and the only way to re-initialize the Gsm Interface is reboot the cpu

pennam, 2024-10-28:
> module already connected
> 
> if you call the GSM.disconnect() function, then the library is not able to recover its initial state and the only way to re-initialize the Gsm Interface is reboot the cpu

@brinata  can you share the sketch to reproduce this scenario?

pennam, 2024-10-28:
@brinata also make sure to call `GSM.end() ` after `GSM.disconnect()` and before the new `GSMbegin()` retry.

brinata, 2024-10-29:
here the sketch [GSMClient.txt](https://github.com/user-attachments/files/17554269/GSMClient.txt)

please note that:
if you run the GSM.begin() with reset = false and you don't call the _device->shutdown() it works
I've modified the end function as follow:
void arduino::GSMClass::end() {
if(_device) {
_device->shutdown();
_device->get_at_handler()->set_urc_handler("^SYSSTART", nullptr);
}
}
because the SYSSTART handler tries to close all the sockets after reboot, locking the modem: cancelling the ^SYSSTART handler the modem is not locket any more, but the connection is still missing ....

... please help

pennam, 2024-10-31:
Hi @brinata 
I was able to reproduce the issue and found a fix. 
I've rebased https://github.com/arduino/ArduinoCore-mbed/pull/975 on the main branch included your suggestion to disable ^SYSSTART urc on shutdown and added the [fix](https://github.com/arduino/ArduinoCore-mbed/pull/975/commits/87c0880fa40c986f491dbae5c153e5d00771337e) that allows reconnections.

I've also updated the connectionHandler PR https://github.com/arduino-libraries/Arduino_ConnectionHandler/pull/131 including a test sketch that is doing the following:
 - connects to the network
 - make recurring http requests while connected
 - disconnect from the network after 60s
 - wait 60s before connecting again to the network and restart http requests

brinata, 2024-11-05:
Hi Pennam,

Was you able to verify if the GSM library works when an "Asynchronous"  disconnection happens ? for instance, when the account credit is expired or when the signal strenght is poor ?

Moreover, very often Network operators allow the connection with no account credit, even if the connection will not work....

how is it possibile to check these condition ?

As writtend in the CINTERION dos, About the ^SYSSTART, I suggest to use it to generate an event that signals the modem startup.

for the shutdown , reset, and re-start , the manual suggest the software fast shutdown command AT^SMSO="fast" to power off the module and the AT+CFUN=1,1 to reset it 

About Hardware reset, the docs say:

**Warning:
Use the EMERG_RST line only when, due to serious problems, the software is not responding
for more than 5 seconds. Pulling the EMERG_RST line causes the loss of all information
stored in the volatile memory. Therefore, this procedure is intended only for use in case of
emergency, e.g. if Cinterion® TX62/TX82 does not respond, if reset or shutdown via AT
command fails.**

brinata, 2024-12-09:
Hi Pennam,

After several tests I discovered that sometimes some modules didn’t reply after a fast startup like after a hardware reset of the EDGE CONTROL board. 
After some more tests I discovered those specific modules re-started replying if left without power supply for at least 60 sec.
After that I discovered that the FAST_SHUTDOWN line of the chip modem is not floating, because is always driver by a level shifter buffer: the net “FST_SHDN_LS” ( 3V3 domain) always drives the net “FST_SHDN” ( 1v8 domain) ;  I tried to place a 10K resistor between the pin 2 - FST_SHDN_LS of the mkr header and the +3V3 and it seems it works better, up to now the catm1 module always starts, in every condition ( the  "FST_SHDN" line seems to be driven by a buffer, non by an open collector circuit )

Is it possible that in some condition the module was kept in permanent shutdown in a "bad" internal state ?

I modified the start-up procedure as follow:

  end();
//  off();

    int attempt = 0;
    while( 1 ) {

        switch(attempt) {
            case 2:
                DEBUG_WARNING("Reset Modem");
                hardwareReset();

            case 1:
                DEBUG_INFO("Starting Modem");

                pinMode(MBED_CONF_GEMALTO_CINTERION_RST, OUTPUT);
                digitalWrite(MBED_CONF_GEMALTO_CINTERION_RST, LOW);

                pinMode(MBED_CONF_GEMALTO_CINTERION_ON, OUTPUT);
                digitalWrite(MBED_CONF_GEMALTO_CINTERION_ON, LOW);  

                pinMode(ON_MKR2, OUTPUT);
                digitalWrite(ON_MKR2, HIGH);

                delay(500);

                on();

                delay(5000);
                break;
            case 0:
                DEBUG_INFO("check Modem");
                break;
            default:
                return 0;
                break;
        }
        if(_device->is_ready() == NSAPI_ERROR_OK) {
            break;
        }
        attempt++;
    }

void arduino::GSMClass::on() {
  /* Module needs a rising edge to power on */
  pinMode(MBED_CONF_GEMALTO_CINTERION_ON, OUTPUT);
  digitalWrite(MBED_CONF_GEMALTO_CINTERION_ON, LOW);
  delay(100);
  digitalWrite(MBED_CONF_GEMALTO_CINTERION_ON, HIGH);
  delay(100);
  digitalWrite(MBED_CONF_GEMALTO_CINTERION_ON, LOW);
  delay(30);
  digitalWrite(MBED_CONF_GEMALTO_CINTERION_ON, HIGH);

}

I added a double power on edge because I found on some Cinterion document that was a workaround to have a sure startup.

The off() call can be useful only if you want to power off the module because this command the module stop communicating 

Can you verify this behaviour on the project documentation  ?

regards

brinata, 2024-12-17:
Hi Pennam,

this morning, at the first starup,  The Portenta CAT.M1/NB IoT GNSS Shield was blocked again ! to unlock it I had to disconnect it from the interposer and then i started working again.....

... Yesterday I didn't disconnect it from edge control, I just switched off the power supply

The fast shutdown pin of the CAT.M1/NB IoT GNSS Shield is still pulled up to 3v3, nevertheless the voltage of the pin was around 1,2 v ...

please help