arduino/ArduinoCore-mbed #971: Arduino Giga Wifi R1: data types limitation on RPC

njadNissi, 2024-10-04:
## Problem case:
Hello guys, I have 10 sensors that I am reading with `analogRead` and storing in `int data[10]` on M4. I need to declare and bind a function on `M4` that can return either a `String` (concatenated values separated by commas) or directly the array of integers onto `M7`. 

## Issue:
 I  have tested all other data types as return types for the function to be bound in `RPC` and apart from `int`, `float` and `char`, any other  is not working. Is there any way to overcome this?

expecting a postive response. thanks

facchinm, 2024-10-04:
Hi @njadNissi , standard c++ containers are supported, so you can push the values in a `std::vector<int>` and use it as return value for the serving RPC function.
Let me know if it works fine :slightly_smiling_face:

njadNissi, 2024-10-06:
> Hi @njadNissi , standard c++ containers are supported, so you can push the values in a `std::vector<int>` and use it as return value for the serving RPC function. Let me know if it works fine 🙂

## Problem: 
Hello, I tried to implement the solution you suggested and I got a bunch of warning and compilation errors.

### My code
[image]

### Compilation output

/home/njad/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/chrono:332:14: note: candidate: template<class _Rep2, class _Period2, class> constexpr std::chrono::duration<_Rep, _Period>::duration(const std::chrono::duration<_Rep, _Period>&)
    constexpr duration(const duration<_Rep2, _Period2>& __d)
              ^~~~~~~~
/home/njad/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/chrono:332:14: note:   template argument deduction/substitution failed:
In file included from /home/njad/.arduino15/packages/arduino/hardware/mbed_giga/4.1.5/cores/arduino/mbed/mbed.h:82:0,
                 from /home/njad/.arduino15/packages/arduino/hardware/mbed_giga/4.1.5/cores/arduino/mbed.h:8,
                 from /home/njad/.arduino15/packages/arduino/hardware/mbed_giga/4.1.5/libraries/RPC/src/RPC_client.h:2,
                 from /home/njad/.arduino15/packages/arduino/hardware/mbed_giga/4.1.5/libraries/RPC/src/RPC.h:14,
                 from /home/njad/Arduino/RPCSensor/RPCSensor_M4/RPCSensor_M4.ino:2:
/home/njad/.arduino15/packages/arduino/hardware/mbed_giga/4.1.5/cores/arduino/mbed/drivers/include/drivers/HighResClock.h:61:72: note:   mismatched types 'const std::chrono::duration<_Rep1, _Period1>' and 'us_timestamp_t {aka long long unsigned int}'
         return time_point{duration{ticker_read_us(get_us_ticker_data())}};
                                                                        ^
In file included from /home/njad/.arduino15/packages/arduino/hardware/mbed_giga/4.1.5/cores/arduino/mbed/rtos/include/rtos/Kernel.h:27:0,
                 from /home/njad/.arduino15/packages/arduino/hardware/mbed_giga/4.1.5/cores/arduino/mbed/rtos/include/rtos/rtos.h:28,
                 from /home/njad/.arduino15/packages/arduino/hardware/mbed_giga/4.1.5/cores/arduino/mbed/mbed.h:24,
                 from /home/njad/.arduino15/packages/arduino/hardware/mbed_giga/4.1.5/cores/arduino/mbed.h:8,
                 from /home/njad/.arduino15/packages/arduino/hardware/mbed_giga/4.1.5/libraries/RPC/src/RPC_client.h:2,
                 from /home/njad/.arduino15/packages/arduino/hardware/mbed_giga/4.1.5/libraries/RPC/src/RPC.h:14,
                 from /home/njad/Arduino/RPCSensor/RPCSensor_M4/RPCSensor_M4.ino:2:
/home/njad/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/chrono:325:23: note: candidate: template<class _Rep2, class> constexpr std::chrono::duration<_Rep, _Period>::duration(const _Rep2&)
    constexpr explicit duration(const _Rep2& __rep)
                       ^~~~~~~~
/home/njad/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/chrono:325:23: note:   template argument deduction/substitution failed:
/home/njad/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/chrono:319:2: note: candidate: constexpr std::chrono::duration<_Rep, _Period>::duration(const std::chrono::duration<_Rep, _Period>&) [with _Rep = long long int; _Period = std::ratio<1, 1000000>]
  duration(const duration&) = default;
  ^~~~~~~~
/home/njad/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/chrono:319:2: note:   no known conversion for argument 1 from 'us_timestamp_t {aka long long unsigned int}' to 'const std::chrono::duration<long long int, std::ratio<1, 1000000> >&'
/home/njad/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/chrono:314:12: note: candidate: std::chrono::duration<_Rep, _Period>::duration() [with _Rep = long long int; _Period = std::ratio<1, 1000000>]
  constexpr duration() = default;
            ^~~~~~~~
/home/njad/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/chrono:314:12: note:   candidate expects 0 arguments, 1 provided
In file included from /home/njad/.arduino15/packages/arduino/hardware/mbed_giga/4.1.5/cores/arduino/mbed/mbed.h:82:0,
                 from /home/njad/.arduino15/packages/arduino/hardware/mbed_giga/4.1.5/cores/arduino/mbed.h:8,
                 from /home/njad/.arduino15/packages/arduino/hardware/mbed_giga/4.1.5/libraries/RPC/src/RPC_client.h:2,
                 from /home/njad/.arduino15/packages/arduino/hardware/mbed_giga/4.1.5/libraries/RPC/src/RPC.h:14,
                 from /home/njad/Arduino/RPCSensor/RPCSensor_M4/RPCSensor_M4.ino:2:
/home/njad/.arduino15/packages/arduino/hardware/mbed_giga/4.1.5/cores/arduino/mbed/drivers/include/drivers/HighResClock.h:61:73: error: no matching function for call to 'std::chrono::time_point<mbed::HighResClock>::time_point(<brace-enclosed initializer list>)'
         return time_point{duration{ticker_read_us(get_us_ticker_data())}};
                                                                         ^
In file included from /home/njad/.arduino15/packages/arduino/hardware/mbed_giga/4.1.5/cores/arduino/mbed/rtos/include/rtos/Kernel.h:27:0,
[... cut ...] e_t&, UnpackVisitor&) [with UnpackVisitor = clmdep_msgpack::v2::detail::create_object_visitor; std::size_t = unsigned int]'
/home/njad/.arduino15/packages/arduino/hardware/mbed_giga/4.1.5/libraries/rpclib/src/rpc/msgpack/v2/unpack.hpp:329:51:   required from here
/home/njad/.arduino15/packages/arduino/hardware/mbed_giga/4.1.5/libraries/rpclib/src/rpc/msgpack/v2/parse.hpp:60:21: warning: left operand of comma operator has no effect [-Wunused-value]

exit status 1

Compilation error: exit status 1

njadNissi, 2024-10-06:
It works just fine. thanks a lot

closed 2024-10-07 by facchinm