arduino/ArduinoCore-mbed #997: Unable to source level debug code in libmbed using IDE 2.3.3.

KurtE, 2024-11-10:
In my case, I am trying to debug some OV5640 camera code I am trying to make work on the GIGA and thought it would be 
easier if I could debug the code using the Arduino IDE debugger.

Took me a while to figure out how to build the debug version of the libmbed.a.  More details about this are in the issue I created
#993 

A lot more details about this is in the forum thread: 
https://forum.arduino.cc/t/building-debug-libmbed-a/1315845/28

My WIP (more like Play in progress) sketch which includes the camera code is up at:
https://github.com/KurtE/Arduino_GIGA-stuff/tree/main/sketches/GigaCameraDisplay_GFX_OV5640
Warning it uses some of the libraries I have in same github project and others.

I am able to start up the debugger and set breakpoints and the like, and get to the line that tries to call into the HAL layer as seen:
[image]

However I can not step into this function, as I am guessing it can not find any source file for it.
That is, the sources I believe in the mbed-os project is contained in the file:
`/home/kurte/github/mbed-os/targets/TARGET_STM/TARGET_STM32F7/STM32Cube_FW/STM32F7xx_HAL_Driver/stm32f7xx_hal_dcmi.c`
Or when generating the debug version of the libmbed.a a similar path in /tmp

I noticed at the end of the mbed-os-to-arduino run there were some copies that were done, I believe from this:
```
copy_core_files () {
	echo -n "Copying generic MbedOS headers to core... "

	rsync -zar --exclude="targets/" --exclude="*TEST*/" --include="*/" --include="*.h"    --exclude="*" \
	mbed-os/ "$ARDUINOCOREMBED"/

	rsync -zar --exclude="targets/" --exclude="*TEST*/" --include="*/" --include="mstd_*" --exclude="*" \
	mbed-os/ "$ARDUINOCOREMBED"/

	echo " done."
}
```
Which makes sense, as the only *.c* file located under the cores/arduino/mbed directory is: mstd_mutex.cpp

Question is: how should it work?
Should I try to copy the .c files into where the header file is?
`/home/kurte/Arduino/hardware/arduino-git/mbed/cores/arduino/mbed/targets/TARGET_STM/TARGET_STM32H7/STM32Cube_FW/STM32H7xx_HAL_Driver/stm32h7xx_hal_dcmi.h
`
Or is it simply not expected to work?

Thanks

Kurt

facchinm, 2024-11-11:
Hi @KurtE ,
to perform source level debug of a rootless .a library, the path of the sources should be added to gdb via [`dir` command](https://sourceware.org/gdb/current/onlinedocs/gdb.html/Source-Path.html) . AFAIK there's no easy way to perform this in IDE 2 debug console, but probably someone from @arduino/team_tooling knows how to modify the `debug.json` file better then 🙂

KurtE, 2024-11-11:
Thanks @facchinm and @umbynos,

I tried a few different directories with the setup I had on my Ubuntu machine.  
The source files are not copied or saved anywhere as part of the mbed-os-to-arduino 
scripts.  

So I tried pointing it to a location where I have done git clone .../mbed-os to my machine. 
I tried doing dir <director with that file> and that did not appear to work.
I tried also adding dir <directory start of mbed-os> 
Also did not find anything.

So, I rebuilt the debug library again following the steps I showed on the forum page, mentioned in the first comment of this issue.

After it completed, I then rebuilt the sketch again, started the debugger, and then opened the debug console and typed in:
`dir /tmp/mged-os-program/mbed-os/targets/TARGET_STM/TARGET_STM32H7/STM32Cube_FW/STM32H7xx_HAL_Driver
`
Assuming I typed this correct.  I am on my main machine right now and looking over at that machine...

And this time it was able to step into the HAL function.  Will be curious to see what happens and I reboot and try again, pointing the debugger at some other directory as the /tmp/mbed-os-program directory will no longer exist.

KurtE, 2024-11-16:
quick update: - maybe information here and thread should be added to added to the debug documentation.

That is, on the same boot you built the debug library, the Arduino IDE will find the sources out in your temp directory.

So: /tmp/mbed-os-program/mbed-os directory to my main hard drive.
Then in subsequent boots, I could get the debugger to work again, by either
a) copy the directory back out to /tmp
b) use the command; set substitute-path <from> <to> command in the debug console.
`set substitute-path /tmp/mbed-os-program /home/kurte/mbed-os-program
` 
Worked on my Ubuntu 24.04 machine as well as my Raspberry PI 5 machine, on the IDE2, I built to run on it.

Today, I was also able to build the debug version of the library on my Windows 11 machine using WSL 24.04, 
I copied both the updated version of the mbed directory to an appropriate place on my windows hard disk.  I also copied the 
/tmp/mbed-os-program/mbed-os to the root of my D drive.

In the debug console, I used the commands: info sources to figure out
what files the debugger was going to look for, and I was able to to update those using the command:

`set substitute-path /tmp/mbed-os-program/./mbed-os d:/mbed_os
`
And I was able to step into the HAL function...
Details of this are up at:
https://forum.arduino.cc/t/building-debug-libmbed-a/1315845/44

@per1234 and all:

Wondering if maybe makes sense to sort of merge this information into my other issue:
#993 

As maybe this is mostly a documentation issue?

It would also be good, if someone would show an example of a debug_custom.json file, that
for example could be used to do one of these set substitute-path commands.
What other information would it need?  Would it be different if I am trying to debug with different debug addapaters?  st-link, versus dap-cmis? 

Thanks all.