arduino/ArduinoCore-mbed #998: Arudino PortentaH7 with the Longan Labs CAN FD Library Porting issue

hpssjellis, 2024-11-12:
Because of my years of work on the [maker100](https://github.com/hpssjellis/maker100) course using the Arduino PortentaH7,  I get a lot of online requests for help. Often I can help by showing my library the [Portenta Pro Community Solutions](https://github.com/hpssjellis/portenta-pro-community-solutions) which can just be searched from the Arduino library manager. But sometimes, like this one, a request for help is way above my abilities. 

@facchinm

@thechordmaster    is trying to connect the PortentaH7 to the automotive CAN FD bus. Here is the link to the library [Longan_CANFD](https://github.com/Longan-Labs/Longan_CANFD). I don't have the hardware so can't do my normal brut force methods, trying stuff until something works or breaks. Any suggestions for  @thechordmaster  about what to try with these examples.

Jeremy Ellis
@hpssjellis 

Having a quick look at the errors after installing the Longan_CANFD library link above it looks like it the issue is about the GPIO pin settings, which is a standard porting portenta issue. Can someone from Arduino have a look at this.

```
C:\Arduino-1.8.19\portable\packages\arduino\hardware\mbed_portenta\4.1.5\cores\arduino/mbed/targets/TARGET_STM/TARGET_STM32H7/STM32Cube_FW/STM32H7xx_HAL_Driver/stm32h7xx_hal_gpio.h:86:36: error: expected identifier before '(' token
 #define GPIO_PIN_0                 ((uint16_t)0x0001)  /* Pin 0 selected    */
                                    ^
C:\arduino-1.8.19\portable\sketchbook\libraries\Longan_CANFD-master\src/mcp2518fd_can_dfs.h:663:16: note: in expansion of macro 'GPIO_PIN_0'
 typedef enum { GPIO_PIN_0, GPIO_PIN_1 } GPIO_PIN_POS;
                ^
C:\Arduino-1.8.19\portable\packages\arduino\hardware\mbed_portenta\4.1.5\cores\arduino/mbed/targets/TARGET_STM/TARGET_STM32H7/STM32Cube_FW/STM32H7xx_HAL_Driver/stm32h7xx_hal_gpio.h:86:36: error: expected '}' before '(' token
 #define GPIO_PIN_0                 ((uint16_t)0x0001)  /* Pin 0 selected    */
                                    ^
C:\arduino-1.8.19\portable\sketchbook\libraries\Longan_CANFD-master\src/mcp2518fd_can_dfs.h:663:16: note: in expansion of macro 'GPIO_PIN_0'
 typedef enum { GPIO_PIN_0, GPIO_PIN_1 } GPIO_PIN_POS;
                ^
C:\Arduino-1.8.19\portable\packages\arduino\hardware\mbed_portenta\4.1.5\cores\arduino/mbed/targets/TARGET_STM/TARGET_STM32H7/STM32Cube_FW/STM32H7xx_HAL_Driver/stm32h7xx_hal_gpio.h:86:47: error: expected ')' before numeric constant
 #define GPIO_PIN_0                 ((uint16_t)0x0001)  /* Pin 0 selected    */
                                               ^
C:\arduino-1.8.19\portable\sketchbook\libraries\Longan_CANFD-master\src/mcp2518fd_can_dfs.h:663:16: note: in expansion of macro 'GPIO_PIN_0'
 typedef enum { GPIO_PIN_0, GPIO_PIN_1 } GPIO_PIN_POS;
                ^
In file included from C:\arduino-1.8.19\portable\sketchbook\libraries\Longan_CANFD-master\src/mcp2518fd_can.h:6:0,
                 from C:\arduino-1.8.19\portable\sketchbook\libraries\Longan_CANFD-master\examples\CAN20_RECV_INT\CAN20_RECV_INT.ino:32:
C:\arduino-1.8.19\portable\sketchbook\libraries\Longan_CANFD-master\src/mcp2518fd_can_dfs.h:663:41: error: 'GPIO_PIN_POS' does not name a type; did you mean 'GPIO_PIN_SET'?
 typedef enum { GPIO_PIN_0, GPIO_PIN_1 } GPIO_PIN_POS;
                                         ^~~~~~~~~~~~
                                         GPIO_PIN_SET
```

facchinm, 2025-02-07:
Hi @hpssjellis ,
sorry for the late response. I don't know if you still need it, but the solution would be to `#undef GPIO_PIN_0` and `#undef GPIO_PIN_1` after including "Arduino.h"

diff --git a/src/mcp2518fd_can_dfs.h b/src/mcp2518fd_can_dfs.h
index 154353f..05bbb8e 100644
--- a/src/mcp2518fd_can_dfs.h
+++ b/src/mcp2518fd_can_dfs.h
@@ -49,6 +49,10 @@ DIRECTLY TO MICROCHIP FOR THIS SOFTWARE.
 #include <inttypes.h>
 #include "mcp_can_fd.h"
 
+#undef GPIO_PIN_0
+#undef GPIO_PIN_1
+#undef CRC
+

Unfortunately the ST headers are quite invasive and "pollute" Arduino.h

hpssjellis, 2025-02-08:
Thanks for looking at it @facchinm. Good suggestion.

thechordmaster, 2025-02-09:
Thank you for clarifying this @facchinm and thank you for pointing me to this thread @hpssjellis . Will try these changes and report back!