arduino/ArduinoCore-mbed #1004: Giga Display going white with LVGL v9.2.2—what is causing this?

littlejohn657, 2024-12-19:
I am having trouble with the screen turning white immediately on startup and not showing anything. I'm using an Arduino Giga with an Arduino Giga Display Shield, LVGL v9.2.2, and Arduino_H7_Video.

To test the display buffer, I replaced the code in the display buffer to colour the screen red (i.e., replaced the code in the second for loop with `Display.set(x, y, 255, 0, 0);`). I saw the red streak through the display in blocks, and then the blocks be trailed almost immediately by black as the red continued down the screen (whether or not I had the code for `lv_obj_t * screen_black = lv_obj_create(NULL);` there or not), then the screen turned white. I also put a serial print in the flush callback that was outputting to the monitor. So I know that the flush callback is being called and is working.

If I replace `lv_obj_t * screen_black = lv_obj_create(NULL);` with `lv_obj_t * screen_black = lv_obj_create(lv_screen_active());`, leave the rest of the code for the screen obj, and remove the code for creating the display and buffers, the screen does appropriately turn black (with a small white border).

Any ideas of what might be going on here?

```c
#include <Arduino_H7_Video.h>
#include <Arduino_GigaDisplay.h> // v1.0.2
#include <Arduino_GigaDisplayTouch.h> // v1.0.1
#include <lvgl.h> // LVGL v9.2.2
#include <ArduinoGraphics.h>

Arduino_H7_Video Display(800, 480, GigaDisplayShield);
Arduino_GigaDisplayTouch TouchDetector;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  
  Display.begin();

  lv_init();

  lv_display_t * display = lv_display_create(800, 480); 

  static uint8_t buf1[800 * 480 / 10];
  lv_display_set_buffers(display, buf1, NULL, sizeof(buf1), LV_DISPLAY_RENDER_MODE_PARTIAL);
  lv_display_set_flush_cb(display, my_disp_flush); 
  lv_disp_set_default(display);

  lv_obj_t * screen_black = lv_obj_create(NULL);
  lv_obj_set_size(screen_black, 800, 480);
  lv_obj_set_style_bg_color(screen_black, lv_color_black(), LV_PART_MAIN);
  lv_obj_set_style_bg_opa(screen_black, LV_OPA_COVER, LV_PART_MAIN);

  lv_screen_load(screen_black);

}

void loop() {
  // put your main code here, to run repeatedly:
  lv_timer_handler();
}

void my_disp_flush(lv_display_t *disp, const lv_area_t *area, uint8_t *color_p) {
  
  // Cast the buffer to lv_color_t for proper interpretation
  lv_color_t *color_map = (lv_color_t *)color_p;

  // Start drawing to the framebuffer
  Display.beginDraw();

  // Loop through the area and set pixels
  for (int32_t y = area->y1; y <= area->y2; y++) {
    for (int32_t x = area->x1; x <= area->x2; x++) {
      // Extract RGB565 from lv_color_t
      uint16_t color = *((uint16_t *)color_map);

      // Convert RGB565 to R, G, B components
      uint8_t r = (color >> 11) & 0x1F;  // Extract 5 bits for red
      uint8_t g = (color >> 5) & 0x3F;   // Extract 6 bits for green
      uint8_t b = color & 0x1F;          // Extract 5 bits for blue

      // Scale to 8-bit values (0-255)
      r = (r * 255) / 31;
      g = (g * 255) / 63;
      b = (b * 255) / 31;

      // Set the pixel color
      Display.set(x, y, r, g, b);

      // Advance to the next color in the buffer
      color_map++;
    }
  }

  // End drawing operation
  Display.endDraw();

  lv_display_flush_ready(disp);
}
```

leonardocavagnis, 2025-01-08:
Hi @littlejohn657,
You don't need to implement the `my_disp_flush` callback in your sketch program, as the library already includes it.

Additionally, the `Display.begin()` function takes care of everything needed to get LVGL running correctly, including LVGL library initialization, buffer setup, and flushing function declaration.

Take a look at this example, [LVGLDemo.ino](https://github.com/arduino/ArduinoCore-mbed/blob/main/libraries/Arduino_H7_Video/examples/LVGLDemo/LVGLDemo.ino), and this [tutorial](https://docs.arduino.cc/tutorials/giga-display-shield/lvgl-guide/) to get started with Arduino_H7_Video and LVGL on the Giga Display Shield.

Here’s a minimal sketch to display a black screen with a "Hello world" label centered on it:
```
#include "Arduino_H7_Video.h"
#include "Arduino_GigaDisplayTouch.h"

#include "lvgl.h" // LVGL v9.2.2

Arduino_H7_Video          Display(800, 480, GigaDisplayShield);
Arduino_GigaDisplayTouch  TouchDetector;

/* "Hello World" example, source: https://docs.lvgl.io/master/examples.html */
void setup() {
  Display.begin();
  TouchDetector.begin();

  /* Change the active screen's background color */
  lv_obj_set_style_bg_color(lv_screen_active(), lv_color_black(), LV_PART_MAIN);

  /* Create a white label, set its text and align it to the center */
  lv_obj_t * label = lv_label_create(lv_screen_active());
  lv_label_set_text(label, "Hello world");
  lv_obj_set_style_text_color(lv_screen_active(), lv_color_hex(0xffffff), LV_PART_MAIN);
  lv_obj_align(label, LV_ALIGN_CENTER, 0, 0);
}

void loop() { 
  lv_timer_handler();
}
```

schnoberts1, 2025-01-10:
I have almost exactly the same issue. 

I get sporadic failures where the device starts and the screen goes white. For me this doesn't happen very often, maybe once every two weeks. When this happens the system continues to run as normal but the display remains white. 

This didn't used to happen. This changed when I fixed an issue in dsi_init where it was supposed to return a value according to the function signature) but it didn't. Since doing this fix (which was absolutely the right thing to do - bugs like this will cause problems with a modern optimiser) I've had the issue. I think my fix just just exposed the bug you are seeing.  

At the point this white screen is happening in my code almost nothing has happened other than it has started up and initialised the display. It may have drawn the first frame of a splash screen but I can't tell :) 

This issue occurs across all the Giga devices. I am currently running a giga restarting through the night with a webcam to see how often it fails so can I can then back out my fix and know with confidence reintroducing the bug in dsi_init solves the problem.

I'm using lvgl 8.

leonardocavagnis, 2025-01-10:
> I have a similar but sporadic issue.
> 
> I get sporadic failures where the device starts and the screen goes white. For me this doesn't happen very often, maybe once every two weeks. When this happens the system continues to run as normal but the display remains white.
> 
> This didn't used to happen. This changed when I fixed an issue in dsi_init where it was supposed to return a value according to the function signature but it didn't. Since doing this fix (which was absolutely the right thing to do - bugs like this will cause problems with a modern optimiser) I've had the issue. I think my fix just just exposed the bug you are seeing.
> 
> At the point this white screen is happening in my code almost nothing has happened other than it has started up and initialised the display. It may have drawn the first frame of a splash screen but I can't tell :)
> 
> This issue occurs across all three Giga devices I have access to. I am currently running a giga restarting through the night with a webcam to see how often it fails so can I can then back out my fix and know with confidence reintroducing the bug in dsi_init solves the problem.
> 
> I'm using lvgl 8.

Hi @schnoberts1,
Thank you for reporting this! 
Your issue appears to be slightly different from the one described here. Please consider opening a new issue (or better, starting a discussion on the [Arduino forum](https://forum.arduino.cc/c/official-hardware/mega/giga-display-shield/191) since the issue is not entirely clear). Include details about the `dsi_init` modifications and any other useful information to help reproduce the problem.

schnoberts1, 2025-01-11:
I have posted here https://forum.arduino.cc/t/random-white-screen-on-start-up/1341499

leonardocavagnis, 2025-04-18:
@schnoberts1 
As mentioned in #1043, this fix (#1050) might help reduce the occurrence of white screens :)

The issue is likely related to the size of the display buffer, try increasing the buffer size and observe whether the white screens disappear.
https://github.com/arduino/ArduinoCore-mbed/blob/f7ff4114bc0d89f23b45423e049f5bc17114dcbc/libraries/Arduino_H7_Video/src/Arduino_H7_Video.cpp#L110
However, be careful: increasing the buffer size might lead to crashes due to high memory usage.

leonardocavagnis, 2025-04-18:
Closing this, as the issue is not caused by the library itself but by incorrect usage of the APIs :)

closed 2025-04-18 by leonardocavagnis