arduino/ArduinoCore-mbed #1005: mbed-os-to-arduino-macos build failure

dirkvdb, 2024-12-21:
I'm trying to build the code because I modified my board configuration to support more sockets.
The build fails however, any ideas what can be the cause of this? I'm using the `gcc-arm-embedded` toolchain from homebrew

```
bash-5.2$ cd ~/.platformio/packages/framework-arduino-mbed/ and bash ./mbed-os-to-arduino-macos -r ~/Projects/mbed-os -a OPTA:OPTA
bash: cd: too many arguments
bash-5.2$ cd ~/.platformio/packages/framework-arduino-mbed/ && bash ./mbed-os-to-arduino-macos -r ~/Projects/mbed-os -a OPTA:OPTA

MBED_CLEAN=0
MBED_UPDATE=0
APPLY_PATCHES=1
RESTORE_GDB_INFO=0
LOCAL_REPO=/Users/dirk/Projects/mbed-os
REMOTE_BRANCH=
MBED_CORE_LOCATION=/Users/dirk/.platformio/packages/framework-arduino-mbed

VARIANT=OPTA BOARD=OPTA
Checking for prerequisites... done.
Creating MbedOS Application... done.
Checking out preferred 'mbed-os' version... Linking local repo /Users/dirk/Projects/mbed-os... done.
Setting up Mbed Application...[mbed] Working path "/private/tmp/mbed-os-program" (program)
[mbed] OPTA now set as default target in program "mbed-os-program"
[mbed] Working path "/private/tmp/mbed-os-program" (program)
[mbed] GCC_ARM now set as default toolchain in program "mbed-os-program"
 done.
Applying patches... done.
Compiling Mbed Application...Compile [  0.1%]: RTX_Config.c
Compile [  0.2%]: irq_cm4f.S
Compile [  0.3%]: cmsis_os1.c
Compile [  0.4%]: rtx_delay.c
Compile [  0.5%]: rtx_lib.c
Compile [  0.7%]: rtx_evr.c
Compile [  0.8%]: rtx_evflags.c
Compile [  0.9%]: rtx_memory.c
Compile [  1.0%]: rtx_kernel.c
Compile [  1.1%]: rtx_mempool.c
Compile [  1.2%]: rtx_mutex.c
Compile [  1.3%]: rtx_msgqueue.c
Compile [  1.4%]: rtx_system.c
Compile [  1.5%]: rtx_semaphore.c
Compile [  1.6%]: os_systick.c
Compile [  1.8%]: mbed_tz_context.c
Compile [  1.9%]: rtx_timer.c
Compile [  2.0%]: rtx_thread.c
Compile [  2.1%]: os_tick_ptim.c
Compile [  2.2%]: mbed_boot_gcc_arm.c
Compile [  2.3%]: mbed_rtos_rtx.c
Compile [  2.4%]: hci_core.c
Compile [  2.5%]: mbed_boot.c
Compile [  2.6%]: main.cpp
Compile [  2.7%]: hci_cmd_bis.c
Compile [  2.9%]: hci_cmd.c
Compile [  3.0%]: hci_cmd_cis.c
Compile [  3.1%]: hci_cmd_ae.c
Compile [  3.2%]: hci_cmd_cte.c
Compile [  3.3%]: mbed_rtx_handlers.c
Compile [  3.4%]: hci_cmd_phy.c
Compile [  3.5%]: hci_cmd_past.c
Compile [  3.6%]: hci_cmd_iso.c
Compile [  3.7%]: hci_vs_ae.c
Compile [  3.8%]: hci_core_ps.c
Compile [  4.0%]: sec_aes.c
Compile [  4.1%]: sec_aes_rev.c
Compile [  4.2%]: sec_ecc_debug.c
Compile [  4.3%]: sec_cmac_hci.c
Compile [  4.4%]: sec_ccm_hci.c
Compile [  4.5%]: sec_ecc_hci.c
Compile [  4.6%]: hci_evt.c
Compile [  4.7%]: sec_main.c
Compile [  4.8%]: att_uuid.c
[... 210 lines cut ...]
       Path: "/private/tmp/mbed-os-program"
       Command: "/Users/dirk/Projects/mbed_env/.pixi/envs/default/bin/python3.13 -u /private/tmp/mbed-os-program/mbed-os/tools/make.py -t GCC_ARM -m OPTA --source . --build ./BUILD/OPTA/GCC_ARM -v"
---
 done.
Generating defines... done.
Generating includes... copying to destination... /tmp/mbed-os-program
 done.
Generating libs..../mbed-os-to-arduino-macos: line 274: ./BUILD/OPTA/GCC_ARM/.link_options.txt: No such file or directory
gcp: cannot stat './BUILD/mbed-core-OPTA.a': No such file or directory
```

This is the output of `/Users/dirk/Projects/mbed_env/.pixi/envs/default/bin/python3.13 -u /private/tmp/mbed-os-program/mbed-os/tools/make.py -t GCC_ARM -m OPTA --source . --build ./BUILD/OPTA/GCC_ARM -v`
```
/Users/dirk/Projects/mbed_env/.pixi/envs/default/bin/python3.13 -u /private/tmp/mbed-os-program/mbed-os/tools/make.py -t GCC_ARM -m OPTA --source . --build ./BUILD/OPTA/GCC_ARM -v
[Warning] @,: Compiler version mismatch: Have 14.2.1; expected version >= 9.0.0 and < 10.0.0
Building project framework-arduino-mbed (OPTA, GCC_ARM)
Scan: framework-arduino-mbed
Using ROM region application in this build.
  Region application: size 0x200000, offset 0x8000000
Using RAM region application_ram in this build.
  Region application_ram: size 0x80000, offset 0x24000000
No Linker Script found

Could not compile for OPTA: No Linker Script found
```

pennam, 2024-12-23:
Hi @dirkvdb if you move to `/tmp/mbed-os-program` and type `mbed compile` you should get a more detailed error

dirkvdb, 2024-12-23:
This is the output of `mbed compile` in `/tmp/mbed-os-program`

```
(mbed_env) bash-5.2$ mbed compile
[mbed] Working path "/private/tmp/mbed-os-program" (program)
[Warning] @,: Compiler version mismatch: Have 14.2.1; expected version >= 9.0.0 and < 10.0.0
Building project mbed-os-program (OPTA, GCC_ARM)
Scan: mbed-os-program
Using ROM region application in this build.
  Region application: size 0x1c0000, offset 0x8040000
Using RAM region application_ram in this build.
  Region application_ram: size 0x80000, offset 0x24000000
Compile [ 29.1%]: sha256_alt.cpp
Compile [ 29.2%]: pn512_irq.c
Compile [ 29.3%]: pn512_hw.c
Compile [ 29.5%]: pn512_cmd.c
Compile [ 29.6%]: PN512TransportDriver.cpp
Compile [ 29.7%]: pn512.c
Compile [ 29.8%]: pn512_registers.c
Compile [ 29.9%]: pn512_timer.c
Compile [ 30.0%]: pn512_poll.c
Compile [ 30.1%]: pn512_rf.c
Compile [ 30.2%]: pn512_transceive.c
Compile [ 30.3%]: stm32h7_eth_init.c
[Error] stm32h7_eth_init.c@69,9: implicit declaration of function 'core_util_critical_section_enter' [-Wimplicit-function-declaration]
[Error] stm32h7_eth_init.c@72,9: implicit declaration of function 'core_util_critical_section_exit' [-Wimplicit-function-declaration]
[ERROR] ./mbed-os/connectivity/drivers/emac/TARGET_STM/TARGET_STM32H7/TARGET_OPTA/stm32h7_eth_init.c: In function 'HAL_ETH_MspInit':
./mbed-os/connectivity/drivers/emac/TARGET_STM/TARGET_STM32H7/TARGET_OPTA/stm32h7_eth_init.c:69:9: error: implicit declaration of function 'core_util_critical_section_enter' [-Wimplicit-function-declaration]
   69 |         core_util_critical_section_enter();
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
./mbed-os/connectivity/drivers/emac/TARGET_STM/TARGET_STM32H7/TARGET_OPTA/stm32h7_eth_init.c:72:9: error: implicit declaration of function 'core_util_critical_section_exit' [-Wimplicit-function-declaration]
   72 |         core_util_critical_section_exit();
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

[mbed] ERROR: "/Users/dirk/Projects/mbed_env/bin/python3.11" returned error.
       Code: 1
       Path: "/private/tmp/mbed-os-program"
       Command: "/Users/dirk/Projects/mbed_env/bin/python3.11 -u /private/tmp/mbed-os-program/mbed-os/tools/make.py -t GCC_ARM -m OPTA --source . --build ./BUILD/OPTA/GCC_ARM"
       Tip: You could retry the last command with "-v" flag for verbose output
---
(mbed_env) bash-5.2$
```

dirkvdb, 2025-01-02:
These are the changes I needed to apply on top of the extrapatches-6.17.0 branch of mbed-os to get it to compile

[mbed-os.patch](https://github.com/user-attachments/files/18293757/mbed-os.patch)