arduino/ArduinoCore-mbed #1011: OPTA / Mbed crash on wifi.begin()

denisperotto, 2025-01-04:
Hi,

I have an issue with my OPTA Wifi board, described here : 

[](https://forum.arduino.cc/t/mbed-crash-on-wifi-begin/1338158)

When i call `wifi.begin()`, the mbed OS crashes (red led blinking 4 times slowly and then fastly). 

Strangely, I have found that writing to the serial port just before invoking `wifi.begin()` seems to avoid the issue, and so this is making the opta examples with WiFi working.

For example, the stock Arduino IDE  "Wifi/WiFiSSLClient" example from the IDE is working correctly, but if I add a :

`delay (1000);` 

just before `wifi.begin()`, mbed OS is crashing.

I though that there was a powering issue, but the same is occuring when I plug the OPTA in a strong USB power supply or 24V industrial supply.

This basic code is crashing the OPTA if place on the setup function, unless I remove the `delay (1000);`  line : 

```
 while (status != WL_CONNECTED)
  {
    Serial.println(ssid); //if not immediately before wifi.begin, mbed will crash
    delay(1000);
    status = WiFi.begin(ssid, pass); 
  }
```

Test conditions : 
- I'm using Arduino IDE 2.3.4. Reported Wifi Firmware is 1.94.0
- I double checked for power supplies issues, same issue is present when powering from the USB port of my computer, an independant USB power supply or a 24V industrial power supply.
- nothing else is connected to OPTA
- if an intentionnaly wrong SSID is used, the OPTA is NOTcrashing. If an intentionnaly wrong PSK is used with a good SSID, the OPTA IS crashing. So I guess it is crashing only when it tries to transmit something

Thanks for reading :-)

Johnson48858, 2025-01-04:
How about trying this:

int status = WiFi.status();
WiFi.begin(ssid, pass); 
 while (status != WL_CONNECTED)
  {
    // your serial print out
    delay(1000);
  }

This works great on my OptaWifi.

denisastro, 2025-01-05:
Hi Kevin

Thanks for your suggestion.

It effectively works, also on mine, apparently thanks to the "int status =
WiFi.status();" line.

If I insert a delay(1000); between that line and wifi.begin, crash again.

In my use case there was something else before wifi.begin, like writing on
digital output.

it seems that there is an incompatibility between some function calls and
wifi.begin, if they are called just before.

I can live with, but I think that there is a bug somewhere.

Johnson48858, 2025-01-05:
Try this instead inside your While statement:  

  while (status != WL_CONNECTED) {
    Serial.print("Attempting to connect to SSID: ");
    Serial.println(ssid);
    status = WiFi.begin(ssid, pass);
    // wait 3 seconds for connection:
    delay(3000);
  }

this part in setup should work for you as well. It works here on mine. This is from the examples for Opta, Under the wifi tag in the ide from the Wifi SSL Client example. I hope this helps you.

denisastro, 2025-01-07:
Thanks @Johnson48858 

this is working as a workaround.

My intention is to help improving the opta product and libraries, so I think It is worth a debug session from the library maintainers (maybe I can help for that).

I have opened a service request to Arduino support, maybe they are able to relay that.

Regards

pennam, 2025-01-13:
@denisastro Thanks for raising this issue. This is a consequence of https://github.com/arduino/ArduinoCore-mbed/issues/1008 and will be fixed by https://github.com/arduino/ArduinoCore-mbed/pull/1012

closed 2025-02-27 by pennam