arduino/ArduinoCore-mbed #1028: Hard fault on Portenta H7 in attachInterrupt() when building with PlatformIO but not with Arduino IDE

unlimitedbacon, 2025-02-22:
I wasn't sure where to post this. Sorry if this is not the right place for PlatformIO issues.

This minimal sketch hard faults immediately (red LED blinking 4 long / 4 short).

```
#include <Arduino.h>
#include <Wire.h>
#include <SPI.h>
#include <Arduino_CAN.h>
#include <ArduinoRS485.h>

volatile unsigned long intCount = 0;

void testInt() {
    intCount++;
}

void setup() {
    pinMode(LED_BUILTIN, OUTPUT); // Use built-in LED for debugging
    digitalWrite(LED_BUILTIN, HIGH); // Turn on LED to indicate setup start

    Serial.begin(115200); // Initialize Serial communication
    while (!Serial) {
        ; // Wait for serial port to connect. Needed for native USB port only
    }
    Serial.println("Serial initialized");

    pinMode(PJ_8, INPUT_PULLUP); // Ensure proper pin configuration
    Serial.println("Pin mode set");

    if (digitalPinToInterrupt(PJ_8) == NOT_AN_INTERRUPT) {
        Serial.println("Pin PJ_8 is not interrupt capable");
    } else {
        attachInterrupt(digitalPinToInterrupt(PJ_8), testInt, RISING); // Attach ISR
        Serial.println("Interrupt attached");
    }

    digitalWrite(LED_BUILTIN, LOW); // Turn off LED to indicate setup end
}

void loop() {
    // Monitor interrupt count
    Serial.print("Interrupt count: ");
    Serial.println(intCount);
    delay(1000); // Add a delay to avoid flooding the serial output
}
```

Interestingly, this happens when building with PlatformIO, but not with the official Arduino IDE. I tried to align the environments as much as possible. On PlatformIO I have the `ststm32` platform version 19.0.0, and `arduino` framework 4.2.1. On the Arduino IDE side I tried with Arduino Mbed OS Portenta Boards version 4.2.1 and 4.2.2.

I was able to set up TRACE32 and track the problem to somewhere within `gpio_irq_init(asm)`. But I have not yet set up source level debugging of Mbed to narrow it down further. Here is the call stack.

```
core_util_critical_section_enter(asm)

gpio_irq_init(asm)

mbed::InterruptIn::InterruptIn(asm)

attachInterruptParam(interruptNum = ?, :func = 0x0804034D, :mode = RISING, :param = 0x0)
  interruptNum = ?
  func = 0x0804034D
  mode = RISING
  param = 0x0
  irq = 0x24016AC0

  }
  mbed::InterruptIn* irq = new mbed::InterruptIn(digitalPinToPinName(interruptNum));
attachInterruptParam(interruptNum = ?, func = 0x0804034D, mode = ?, param = 0x0)
  interruptNum = ?
  func = 0x0804034D
  mode = ?
  param = 0x0

    attachInterruptParam(PinNameToIndex(interruptNum), func, mode, param);
attachInterrupt(:interruptNum = PJ_8, :func = 0x0804034D, :mode = RISING)
  interruptNum = PJ_8
  func = 0x0804034D
  mode = RISING

  attachInterruptParam(interruptNum, (voidFuncPtrParam)func, mode, NULL);
setup()

    if (digitalPinToInterrupt(PJ_8) == NOT_AN_INTERRUPT) {
        Serial.println("Pin PJ_8 is not interrupt capable");
    } else {
        attachInterrupt(digitalPinToInterrupt(PJ_8), testInt, RISING); // Attach ISR
main()

#endif

        setup();
osThreadExit(asm)

end of frame

```

manchoz, 2025-02-24:
Hey @unlimitedbacon,
Unfortunately, we don't support using PlatformIO on Arduino Boards. We know there have been and still are several issues with the support of MbedOS-based boards on PIO, so we strongly recommend switching back to the plain Arduino IDE.

unlimitedbacon, 2025-02-24:
Understood. Is this the right place to report this problem or should I post it elsewhere?

manchoz, 2025-02-24:
I really don't know. Maybe on the Platform IO repo or forum?

I don't know if they have other channels for support.

closed 2025-05-06 by manchoz