arduino/ArduinoCore-mbed #1042: LittleFileSystem2 crashes in unmount in some scenarios

schnoberts1, 2025-03-26:
Consider:

```
 auto& userPartition { qspi.userPartition() }; // This returns a reference to a block device.
 LittleFileSystem2 userFilesystem( "user" );
 auto err = userFilesystem.reformat( &userPartition );
```

This crashes (depending on where it is in the code base) because reformat calls unmount. Unmount checks whether userFilesystem._bd is null and if not tries to deinit() it. Sadly the constructor for LittleFileSystem2 does not initialise _bd so it's whatever is lying around. LittleFileSystem doesn't have this issue as it initialises its fields properly.

I know this is a dead ARM port that the core just uses, but I spent an hour tracking it down so I thought I'd log it for others to find.

multiplemonomials, 2025-04-15:
Hi! Nice find on this, I created a PR to fix this bug on Mbed Community Edition! https://github.com/mbed-ce/mbed-os/pull/451

pennam, 2025-05-05:
Thanks @multiplemonomials. I've opened a [PR](https://github.com/arduino/ArduinoCore-mbed/pull/1059) to include your fix in the next core release