arduino/ArduinoCore-mbed #1043: Arduino H7 Video - LVGL9.X Double buffer is allocated but never used

thodorii, 2025-03-27:
Hi, I've been running into what seems to be memory issues causing an MBED OS crash with LVGL. I was looking around to see draw buffer allocations and ran into this code. I am confused on the purpose of the 2nd draw buffer malloc when it is never used, is this an oversight?

 https://github.com/arduino/ArduinoCore-mbed/blob/f7ff4114bc0d89f23b45423e049f5bc17114dcbc/libraries/Arduino_H7_Video/src/Arduino_H7_Video.cpp#L108-L130

thodorii, 2025-04-07:
I can confirm that after disabling the second buffer, I no longer see the MBED OS crashes I used to consistently get on **lv_timer_handler().**

leonardocavagnis, 2025-04-18:
Hi @thodorii
Yes, leaving there `buf2` is probably an oversight.
Thanks for pointing it out!

But, I can't replicate the MBED OS crash. Please, could you post the sketch causing this?

thodorii, 2025-04-18:
Unfortunately I don't think I can share the sketch due to it being part of work.

But the crash is likely due to memory constraints and the double buffer was the thing putting it in crashing territory.

The MBED OS crashes were happening with lv_timer_handler() and I was never able to get an exact trace of where it crashed.

leonardocavagnis, 2025-04-18:
Ok, no problem! :)
I opened a PR with the suggested fix!

By the way, I decided not to increase the buffer size because, as suggested in the LVGL documentation, `LV_DISPLAY_RENDER_MODE_PARTIAL` can work with a buffer smaller than the display size.
I’m aware this could cause some stability issues, but increasing the buffer size would consume too much memory and lead to crashes.

thodorii, 2025-04-18:
Ok great! 

Anecdotally I also started seeing far fewer white screens after removing the buffer #1004 , although they were not completely gone.

PR #1050 (2025-04-18): Arduino H7 Video + LVGL 9.x: Remove unused buffer
Fix #1043
closed 2025-04-18 by facchinm