arduino/ArduinoCore-mbed #1048: Arduino H7 Video - Display remains black when executed on the M4 processor

MikeMM101, 2025-04-11:
I have found a bug in the class Arduino_H7_Video. When creating an object with the call
"Arduino_H7_Video Display(480, 800, GigaDisplayShield); "
this works with the M7 processor, but not with the M4 processor. The display remains black. The call only works on the M4 processor if I swap the values 480 and 800. But then everything on the display appears rotated by 90 degrees and a rotation with the function lv_display_set_rotation(...) does not work.

I use Arduino Giga R1 with lvgl 9.2.2 and Mbed OS 4.2.4

This Example Code should display a Button:

`#include <mbed.h>
#include "RPC.h"

#ifdef CORE_CM7
void setupM7() {
  RPC.begin();
  Serial.println("setupM7()");
  delay(100);
}

void loopM7() {
  Serial.println("loopM7()");
  delay(1000);
}
#else
#define LV_CONF_INCLUDE_SIMPLE 1

#include <lvgl.h>
#include <Arduino_GigaDisplay.h>
#include <Arduino_GigaDisplayTouch.h>
#include "Arduino_H7_Video.h"

lv_obj_t* g_pObjButton = NULL, *g_pObjBtnLabel = NULL, *g_pWLANScreen = NULL;

// This doesnt work on M4
// Arduino_H7_Video          Display(800, 480, GigaDisplayShield);

// This works on M4, but the Dispaly is rotated by 90 degrees
Arduino_H7_Video          Display(480, 800, GigaDisplayShield);

Arduino_GigaDisplayTouch  TouchDetector;
GigaDisplayRGB rgb;

static void btn_event_cb1(lv_event_t* e) {
  //Serial.println("btn_event_cb1");
}

void initScanButton() {
  g_pWLANScreen = lv_obj_create(lv_scr_act());
	lv_obj_set_size(g_pWLANScreen, Display.width(), Display.height());
	lv_obj_clear_flag(g_pWLANScreen, LV_OBJ_FLAG_SCROLLABLE);
	lv_obj_center(g_pWLANScreen);

	g_pObjButton = lv_btn_create(g_pWLANScreen);
	lv_obj_set_width(g_pObjButton, 200);
	lv_obj_set_height(g_pObjButton, 80);
	lv_obj_set_x(g_pObjButton, 0);
	lv_obj_set_y(g_pObjButton, 0);
	lv_obj_add_event_cb(g_pObjButton, btn_event_cb1, LV_EVENT_RELEASED, NULL);

	lv_obj_clear_flag(g_pObjButton, LV_OBJ_FLAG_SCROLLABLE);
	lv_obj_set_align(g_pObjButton, LV_ALIGN_CENTER);
	lv_obj_add_flag(g_pObjButton, LV_OBJ_FLAG_SCROLL_ON_FOCUS);     /// Flags
	lv_obj_clear_flag(g_pObjButton, LV_OBJ_FLAG_SCROLLABLE);      /// Flags
	lv_obj_set_style_bg_color(g_pObjButton, lv_color_hex(0xDCDCDC), LV_PART_MAIN | LV_STATE_DEFAULT);
	lv_obj_set_style_bg_opa(g_pObjButton, 255, LV_PART_MAIN | LV_STATE_DEFAULT);
	lv_obj_set_style_border_color(g_pObjButton, lv_color_hex(0xB4B4B4), LV_PART_MAIN | LV_STATE_DEFAULT);
	lv_obj_set_style_border_opa(g_pObjButton, 255, LV_PART_MAIN | LV_STATE_DEFAULT);
	lv_obj_set_style_border_width(g_pObjButton, 2, LV_PART_MAIN | LV_STATE_DEFAULT);
	lv_obj_set_style_text_color(g_pObjButton, lv_color_hex(0x808080), LV_PART_MAIN | LV_STATE_DEFAULT);
	lv_obj_set_style_text_opa(g_pObjButton, 255, LV_PART_MAIN | LV_STATE_DEFAULT);
	lv_obj_set_style_text_align(g_pObjButton, LV_TEXT_ALIGN_CENTER, LV_PART_MAIN | LV_STATE_DEFAULT);
	lv_obj_set_style_text_font(g_pObjButton, &lv_font_montserrat_14, LV_PART_MAIN | LV_STATE_DEFAULT);
	g_pObjBtnLabel = lv_label_create(g_pObjButton);
	lv_label_set_text(g_pObjBtnLabel, "Scannen");
	lv_obj_set_style_text_color(g_pObjBtnLabel, lv_color_black(), 0);
	lv_obj_center(g_pObjBtnLabel);
}

void setupM4() {
  RPC.begin();
  Display.begin();
	TouchDetector.begin();
	rgb.begin();

  initScanButton();

}

void loopM4() {
  lv_timer_handler();

}
#endif

void setup() {
#ifdef CORE_CM7
	setupM7();
#else
	setupM4();
#endif
}

void loop() {
#ifdef CORE_CM7
	loopM7();
#else
	loopM4();
#endif

}`

leonardocavagnis, 2025-04-23:
Hi @MikeMM101,
Thanks for reporting this!
Try downgrading the LVGL library to version 8.x.x, that should get things working.
In the meantime, I’m investigating the compatibility issues with version 9.x.x.

leonardocavagnis, 2025-04-23:
Hey, 
I probably found the error!
Give a spin with this fix #1052 and let me know :)
Thanks