arduino/ArduinoCore-mbed #1079: analogWrite does not work well with Arduino RP2040 board

fc-projects, 2025-07-21:
Hi!

I wanted to do something really simple. Having several LEDs controlled by PWM.
Everything was working very weird with analogWrite so I tried several boards and reduced the setup to the minimal.
It is really easy to reproduce, just put one LED on D8 and one another on D12 with appropriate pull-down resistors.
Use this simple two lines of code into the setup function which only use analogWrite to lit up the LEDs to independant values.
`analogWrite(8, 255);`
`analogWrite(12, 0);`
and
`analogWrite(8, 0);`
 `analogWrite(12, 255);`

When applying those PWM, the last one is always taken for both LEDs.
It behaves exactly like there was an overwrite of the first one by the second one.
Using only digitalWrite works well to lit up one LED and not the other one and vice versa but if one analogWrite have been used digitalWrite does not work anymore.
I checked the schematics several times, there is no reason why it would do that in hardware.
I believe this is a software bug, but I was not able to find where. I am running with the last board package 4.3.1.
It seems there is the same bug with D17, D18 & D19 while those last ones have pull-ups because they are used by I²C this may explain the link between D18 & D19 but not D17!

megacct, 2025-07-21:
Hi, out of interest, do you get the same result specifying D8 and D12 in the analogwrite calls?

fc-projects, 2025-07-21:
Just tested with p20 & p4, same behaviour.

megacct, 2025-07-21:
over to the Arduino team then, does feel like a bug rather than just a mapping prob.